MarlinFirmware/Marlin #16583: [BUG] (Cannot Compile for LPC1768 or LPC1769)

nmthastings, 2020-01-16:
### Bug Description

With all currently hosted versions of Marlin 2.0 (versions 2.0.x, bugfix-2.0.x, and dev-2.1.x, located at **http://marlinfw.org/meta/download/**), Marlin cannot compile for LPC1768 or LPC1769.

### My Configurations
I have included the entirety of the Marlin project below. I am using Win10, VSCode 1.4.10, and the latest version of platformio. 

My printer configuration is irrelevant at this point because I make no changes to config or config_adv besides those listed below, however my printer is an ender 3 with a new Big Tree Tech SKR V1.4 Turbo with a RPi 3B controller running octoprint.

All changes are made to a brand new, freshly downloaded copy of Marlin.

[Marlin-dev-2.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/4068991/Marlin-dev-2.1.x.zip)

### Steps to Reproduce

1. Open platformio project.
2. Edit line 21 in platformio.ini to 'default_envs=LPC1769' or 'default_envs=LPC1768'
3. Edit line 133 in configuration.h to '#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_4_TURBO' 
4. Build

###Expected Behavior:

A perfectly built firmware that I can use to deposit melted plastic accurately on a build surface.

###Actual Behavior:

Building fails due to errors called in several files with an increasingly annoying error message:

'error "Oops! Set MOTHERBOARD to an LPC1768-based board when building for LPC1768."'
(NOTE: the error specifies LPC1768 regardless of the value of default_env in platformio.ini; That is to say, it says LPC1768 if 'default_env=LPC1768' or if 'default_env=LPC1769'. And BOARD_BIGTREE_SKR_V1_4_TURBO is an LPC1769 board.)

#### Additional Information

I originally thought it was an issue with BOARD_BIGTREE_SKR_V1_4_TURBO since it is relatively new, however I attempted the same build with BOARD_BIGTREE_SKR_V1_3 and had the same errors. As I said, I have attempted this with every version of Marlin 2.0 available at the moment.

I want to turn my printer into an anet A8 so it spontaneously combusts at this point. Please github, you are my only hope.

thisiskeithb, 2020-01-16:
The config files in the attached zip have `BOARD_RAMPS_14_EFB` defined as the `MOTHERBOARD`. Do you have another zip? You only need to attach the config files, not the entire Marlin project.

sjasonsmith, 2020-01-16:
> The config files in the attached zip have `BOARD_RAMPS_14_EFB` defined as the `MOTHERBOARD`. Do you have another zip? You only need to attach the config files, not the entire Marlin project.

Actually,I suspect that MOTHERBOARD setting is the entire problem.

nmthastings, 2020-01-16:
> 
> 
> The config files in the attached zip have `BOARD_RAMPS_14_EFB` defined as the `MOTHERBOARD`. Do you have another zip? You only need to attach the config files, not the entire Marlin project.
> 
> Edit: I just rebased [my SKR 1.4 config](https://github.com/thisiskeithb/Marlin/tree/myskr) on top of the latest commit ([c305c2c](https://github.com/MarlinFirmware/Marlin/commit/c305c2cc6bb71283bf675e92c120aa99ceb030b5)) and it compiles fine.

...........I've been editing the wrong config file. 

I'm going to take up drinking.

sjasonsmith, 2020-01-16:
I think we've all done it at one point!

> Fell free to close the issue.

Most of us can't actually close out issues. Would you mind closing it yourself, @nmthastings? You always have the power to close the issues you've created.

nmthastings, 2020-01-16:
> 
> 
> I think we've all done it at one point!
> 
> > Fell free to close the issue.
> 
> Most of us can't actually close out issues. Would you mind closing it yourself, @nmthastings? You always have the power to close the issues you've created.

I literally got so excited to dive back into the firmware that I missed the 'Close and Comment' button. Wow today has been a day.

closed 2020-01-16 by nmthastings