MarlinFirmware/Marlin #16585: [BUG] Linear advance break filament loading

drewmoseley, 2020-01-16:
### Bug Description

I have a 2.0.1 setup for an Ender 3.  I have tweaked the load/unload settings so that I am able to properly change the filament using the menu options.  When I enable linear advance, the unload sequence works properly however the load does not move the extruder motor at all.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Configure Marlin for proper load/unload with LIN_ADVANCE disabled
2. Verify that it works
3. Enable LIN_ADVANCE
4. Try to "CHANGE FILAMENT"

**Expected behavior:** [What you expect to happen]

Filament is unloaded.
New filament is loaded.

**Actual behavior:** [What actually happens]

Filament is unloaded.
No new filament is loaded. No extruder motor activity.

#### Additional Information

drewmoseley, 2020-01-16:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4073393/configs.zip)

randellhodges, 2020-01-16:
I noticed you have an E driver of `TMC2208_STANDALONE`. Are these in stealthchop or spreadcycle mode? There are issues with linear advance with TMC2208s in stealthchop.

Vertabreak, 2020-01-16:
i have tmc2208 on XYZE in standalone the filament load/unload feature works normal on my printer.

drewmoseley, 2020-01-16:
I have the Creality silent mainboard. Honestly I'm not sure whether the steppers are stealthchop or spreadcycle.

There's no obvious obstruction or voltage issue since it works when LIN_ADVANCE is commented out.

Vertabreak, 2020-01-16:
suspecting hardware defect there are a number of stealthchop V1 tmc2208s that have a ton of issues.
i have the newer stealth chop V2 ones.

boelle, 2020-01-21:
i also have LA enabled and no issue during filament loading

but i dont use 2.0.1 but bugfix 2.0.x

boelle, 2020-01-26:
closing since this is most likely a hardware issue

we can reopen if that is not the case

closed 2020-01-26 by boelle