MarlinFirmware/Marlin #16588: [FR] Should `WIFI_SSID` and `WIFI_PWD` etc. be moved to e.g. `Configuration_secrets.h`?

felixstorm, 2020-01-17:
### Description
As WiFi on the ESP32 now gets more and more functional, it feels a bit unpleasant to put my WiFi credentials inside `Configuration_adv.h` and then to accidentally push them to my public Marlin repo on GitHub.

My suggestion would be to put them inside a separate config file which only gets included if needed (so the file would not even have to exist for users not using WiFi etc.). Users could then either ignore it completely in Git (or mark as `skip-worktree`) to not accidentally publish their secrets. I did *not* choose the name `Configuration_WiFi.h` but `Configuration_secrets.h` as I already saw that there might be more secrets to come, e.g. website authentication credentials (https://github.com/luc-github/ESP3DLib/commit/8fc6354e5a448a06624b90fe72ecb52626613307) etc.

I will be happy to provide a corresponding PR, I just wanted to check the general opinion on this first before investing more time on it.

vivian-ng, 2020-01-17:
@felixstorm If you are using the ESP3DLib, I think @luc-github has made it such that you can set up the WiFi credentials over the webUI. If you compile with the default SSID and password in Configuration_adv.h, ESP3DLib should ignore them, and try to see if any credentials are stored in SPIFFs. If so, it will use those; if not, it will start up an AP. You can then connect to the AP, and configure the WiFi credentials using the webUI. @luc-github can probably share more about how it actually works.

luc-github, 2020-01-17:
WiFi settings can be ignored with ESP3DLib, they can be defined for convenience but they are not mandatory

boelle, 2020-01-17:
i have rephrashed the title as this is a FR and the title suggested it was a question, the former is allowed on the issue list the later is not

felixstorm, 2020-01-18:
@vivian-ng, @luc-github Yes, you are right and I am aware that you do not *need* to use the defines in `Configuration_adv.h` for ESP3DLib but can set up the credentials in the AP web UI.

But I would still consider it bad practice to encourage users to potentially put secret information inside a file that may get pushed to their public GitHub repository.

vivian-ng, 2020-01-19:
@felixstorm Maybe the current WiFi portion can be commented to:
```
// To use your own WiFi SSID and password, uncomment the following line:
//#include "credentials.h"
// To set WiFi SSID and password, put the following to lines into a file called 'credentials.h',
// and replace 'Wifi SSID' and 'Wifi Password' with your own respective SSID and password.
//#define WIFI_SSID "Wifi SSID"
//#define WIFI_PWD  "Wifi Password"
```

Then, add `credentials.h` to `.gitignore` so that it does not get pushed.

What do you think?

felixstorm, 2020-01-24:
@vivian-ng Yes, that sounds even better than trying to provide an almost blank file to everybody - I will probably post a corresponding PR within the next few days.

closed 2020-06-30 by github-actions[bot]