MarlinFirmware/Marlin #16590: [BUG] AUTO_BED_LEVELING_BILINEAR PROBE_MANUALLY crashes after last point

Golkun, 2020-01-17:
### Bug Description

AUTO_BED_LEVELING_BILINEAR PROBE_MANUALLY crashes after last point

### My Configurations
Marlin 2.0.1 from 24.12.2019
corexy
inductive probe

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
Marlin.config.zip attached
[20200117_171546_97394802255645.zip](https://github.com/MarlinFirmware/Marlin/files/4077690/20200117_171546_97394802255645.zip)

### Steps to Reproduce
1. Select Auto Bed Leveling from the LCD menu
2. go through the points manually
3. after last point has been finished a click on the encoder forces a reset

**Expected behavior:** no reset and values recorded to be stored in eeprom

**Actual behavior:** a reset occurs, not able to finish auto bed leveling.

[Marlin config.zip](https://github.com/MarlinFirmware/Marlin/files/4077657/Marlin.config.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-01-17:
is it the same in bugfix 2.0.x ? that is newer than 2.0.1

Golkun, 2020-01-17:
Will try bugfix 2.0.x

Golkun, 2020-01-17:
can I use configuration.h and configuration_adv.h from my current build?

boelle, 2020-01-17:
not sure, best is to compare them with a tool like winmerge

Golkun, 2020-01-18:
Btw Winmerge is a great tool.
Unfortunately the behaviour is still the same with Bugfix 2.0.x

boelle, 2020-01-23:
how often do you update? bugfix is updated almost daily

boelle, 2020-01-30:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-30 by boelle