MarlinFirmware/Marlin #16594: [FR] Unretract speed in linear advance calibration pat

josedpedroso, 2020-01-18:
### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

memphisanid, 2020-01-18:
If you change the retract speed to 25, does it do the same thing in the gcode generator?

josedpedroso, 2020-01-18:
Setting retract speed to 25 does solve the problem, though I get a little bit of stringing between some lines. However, to me, that's not the issue. The problem is that by mindlessly plugging known good values from my slicer into the page, I was getting misleading results. I thought my printer was, for some reason, simply incapable of handling higher k-factors. It's just an Anet A8, after all.

closed 2020-06-30 by github-actions[bot]