MarlinFirmware/Marlin #16604: [BUG] Stepper movement broken on BIGTREE_SKR_PRO

teeminus, 2020-01-18:
Dear all.

### Bug Description

I checked out the latest revision of the bugfix-2.0.x branch (80400ab) and noticed, that the steppers won't move on my SKR Pro board. Instead of moving, the steppers some high frequency noice and then the printer halts.

### My Configurations
SKR Pro v1.1
TMC5160 for all steppers
CoreXY kinematics
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4080933/config.zip)

### Steps to Reproduce

1. Update firmware
2. Auto home printer

**Expected behavior:** Printer homes.

**Actual behavior:** Stepper motors make high frequency noice and the printer halts.

#### Additional Information
The bug was introduced when merging #16452 (1ad53cee). I traced the bug down to the file Marlin/src/HAL/HAL_STM32/timers.h:65: `#define STEP_TIMER 9  // STM32F401 has no TIM6, TIM7, or TIM8`
When I change this line to `#define STEP_TIMER 6` as it was before, the printer works fine.

I didn't create a PR because I did'nt know if this issue has to be fixed in that file or in the pins.h for the SKR Pro. Based on the comment of that line, fixing this bug for me might break other printers.

sjasonsmith, 2020-01-18:
I was worried those timer changes might cause issues, but I wasn’t in a position to test it when I saw the PR. The timer capabilities vary a lot between STM32 chips, and a better way to deal with that is needed.

In addition to the timers in timers.h, additional timer are assigned in the variants file and still more by the framework depending on the exact features enabled on the board. It is very easy to cause conflicts.

Maybe this HAL should just require these to all be defined in a variants file for every board.

Minims, 2020-01-18:
OK same issue here with TMC2209 UART. I though I have broken my board :-). Back to 2.0.1 and it works for now.

viper93458, 2020-01-18:
Same issue for me with 2208s.

-William

peeff, 2020-01-19:
The same problem TMC2130_STANDALONE

Hans007a, 2020-01-19:
I saw this change and this made a problem with 2208, and I set it back to 6.

BastR, 2020-01-19:
Same here with TMC5160

thisiskeithb, 2020-01-20:
> The bug was introduced when merging #16452 (1ad53ce). I traced the bug down to the file Marlin/src/HAL/HAL_STM32/timers.h:65: #define STEP_TIMER 9 // STM32F401 has no TIM6, TIM7, or TIM8
> When I change this line to #define STEP_TIMER 6 as it was before, the printer works fine.

We're also seeing some crazy fast movements with TMC2209s on BigTreeTech's BTT002 board (`STM32F407VET6`). I thought I screwed up a config setting, but reverting the timer fixed the issue.

Bob-the-Kuhn, 2020-01-20:
Rattsss!!!!  

Are all the problems on non 401 boards?

Could a few of you see if you can come up with a set of timers that works for the 407 chips from this list?
`1-5, 9-11`

The timers I know of are:
```
TIMER_SERVO 
TIMER_TONE
STEP_TIMER
TEMP_TIMER 
```

I'm going to test some macros I've found to see if **#if defined(..)** statements will allow us to independently set the 401 timers from all the other STM32F4xx devices.

sjasonsmith, 2020-01-20:
> The timers I know of are:

There is also  TIMER_SERIAL, which is defined at least in the SKR Pro variant file.

sjasonsmith, 2020-01-20:
> > The timers I know of are:
> 
> There is also TIMER_SERIAL, which is defined at least in the SKR Pro variant file.

If you don't specify TIMER_SERIAL and use UART-connected TMC drivers, the code in the STM32 framework will automatically select what it decides is the best timer. This may or may not actually be the best timer for the board.

If the board could potentially be used with UART-connected TMC drivers, it would be best to explicitly declare TIMER_SERIAL in the variant file.

Bob-the-Kuhn, 2020-01-20:
Please take a look at PR #16621.  I think this will make both the STM32F401 and the STM32F407 boards happy.

teeminus, 2020-01-20:
I think so too. I will try the fix when I am back home.

teeminus, 2020-01-20:
The PR fixes the issue on my SKR Pro (STM32F407_5ZX).

closed 2020-01-21 by thinkyhead