MarlinFirmware/Marlin #16605: No USB Connection with SKR PRO

kurtisane, 2020-01-18:
Hi,
I'm struggeling the whole day already to get the SKR PRO v1.1 working again. I had a certain version of this repo with my configs. Just wanted to change a Value on STEPS_PER_UNIT and compile and put it on the printer. Now I'm not able to connect anymore and when the Board is powered via my powersupply Windows is not able to recognize the board.
Serial Port is set to -1, so it should tell Marlin to use the USB Port. 

Can someone upload a marlin project (with firmware.bin, config, etc) that is working for him with USB Port. I just want to know if my board is somehow bricked or not. 

Any one else facing issues getting the SKR Pro to run ?

teeminus, 2020-01-18:
Which branch of Marlin are you using? If possible, please provide the revision of that branch.

You can check the config from my issue (#16604). For me, the USB port is working and I can use it the send GCODE to my printer.

Otherwise please upload your config. I could check for config and see if I can spot what might causes the problem.

boelle, 2020-01-18:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

kaibs, 2020-01-20:
Quoting myself from an [Issue on the BTT SKR](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/197#issuecomment-576146022) repo:

_This issue might be caused by a damaged thermistor (or probably endstop). For the SKR v1.3 and the v1.4 a shorted sensor seems to crash Marlin and additionally somehow kills the boards USB-connection._

It would be interesting if this is the case for the SKR Pro too, or even a problem (or feature?) in Marlin itself.

boelle, 2020-01-21:
closing this one until we have the configs as required by the issue template

please add @ in front of my username once it has been added

closed 2020-01-21 by boelle