MarlinFirmware/Marlin #16607: framework-arduinoststm32 problem?

Bob-the-Kuhn, 2020-01-18:
There may be a problem with one of the versions of platformio's **framework-arduinoststm32**.

I ran into a failure of a macro so I created PR #16589 to fix it.   After that I ran into some other strange compile issues so I deleted  the **.platformio\packages\framework-arduinoststm32** directory and did a clean compile.  All the problems were gone after that.

Now I see lots of PRs failing a test as a result of a macro problem on a STM32 board.

Could they be related?

Vertabreak, 2020-01-18:
seems like a reasonable conclusion i would also suspect the issues to be linked in some way.

sjasonsmith, 2020-01-19:
I am able to build for boards using both the STM32 and STM32F1 HALs. Perhaps somebody experiencing the issue could post their configs, along with the section at the beginning of the build output that shows all the dependency versions used.

jedld, 2020-01-19:
Experiencing a similar problem, but with an skr 1.3 board. This happens if I enable  TMC2209 driver type. Compiles fine on the default A4988 driver type.

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.6.1
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfigPre.h:54,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../core/drivers.h:71:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../core/drivers.h:106:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
```

CRCinAU, 2020-01-19:
Same on an SKR E3 Mini v1.2:

```
HEAD is now at 23fc81f1a [cron] Bump distribution date (2020-01-19)
Processing STM32F103RC_bigtree_512K_USB (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.0.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: framework-arduinoststm32-maple 1.10000.190819 (1.0.0), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-stm32duino 1.0.1
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal>
|-- <TMCStepper> 0.6.1
|   |-- <SoftwareSerialM>
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <Adafruit MAX31865 library> 1.0.3
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SoftwareSerialM>
|-- <EEPROM>
|-- <STM32ADC> 1.0
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/STM32F103RC_bigtree_512K_USB/src/src/HAL/HAL_STM32F1/HAL.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_512K_USB/src/src/HAL/HAL_STM32F1/HAL_SPI.cpp.o
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfigPre.h:54:0,
                 from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:71:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:106:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/../../Configuration_adv.h:2025:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(Z4)
       ^~~~~~~~~~~
```

sjasonsmith, 2020-01-19:
Thanks to those of you who posted your build output.
We discovered the issue causing those exact errors, there is more discussion in issue #16611 

It's not clear to me whether @Bob-the-Kuhn's original issue is directly related to this or not. We would need to see if more people encounter it _without_ it being problems with example files, and can provide their build output.

Bob-the-Kuhn, 2020-01-19:
The platform in question is ST's STM32 library which Marlin uses for the STM32F4xx and STM32F7xx boards.  The STM32F1xx boards use a different platform/library.

sjasonsmith, 2020-01-19:
I think all current PR failures are due to the problem with the examples. It will impact all platforms, it just seems that STM32F1 tests are often the fastest to report the failure and the rest get cancelled.

It seems that any test that uses an example config and enabled TMC drivers will fail due to the issue described in #16611.

@Bob-the-Kuhn, do you think there is still a separate issue impacting the STM32 HAL? Do you have a link to any failures which are _not_ related to the `AXIS_IS_TMC` macros?

Bob-the-Kuhn, 2020-01-20:
I didn't save my failure log and I can't get back to the failing platform so I can't provide details beyond what is in PR #16587.

Bob-the-Kuhn, 2020-01-20:
No one else is seeing this issue so it  probably was just a quirk in my PC.

closed 2020-01-20 by Bob-the-Kuhn