MarlinFirmware/Marlin #16610: Print finish the file quickly

ColdeZhang, 2020-01-19:
When the machine finish the heating, It read the file very quickly, the stepper even didn't start to move the print job had finished.
I used computer to log the communication. I found, what ever send to the printer, It replied "OK" immediately. I supposed, It should reply "bussing". 
When I operate It manually It works properly. 
I don't know what happened. I have tried to use a another slicer software, but It doesn't work. So I suspect the problem is about the firmware, and looking for help here.
By the way, the printer just been assembly and haven't work before. It's a Kossel, and the version of firmware is 1.1.3.

ellensp, 2020-01-19:
Sounds like your uploading to the printer vs printing to it. So your just copying the gcode file to the printer vs running it. Secondly we can only help with current versions of firmware. The first response will likely be upgrade your firmware, see if that helps. Thirdly you disregarded the instruction and didn't provide configuration files, makes me wonder what other instructions you failed to follow when building your machine...

ColdeZhang, 2020-01-19:
Well actually, i just found the problem. The default setting of slicer origin is corner, while my printer is centre, so when I slice the model the actually print location is out of the setting in firmware. So Marlin skip such G_Code.

closed 2020-01-19 by ColdeZhang