MarlinFirmware/Marlin #16611: bugfix-2.0.x fails to compile skr e3 mini 1.2

carterdo, 2020-01-19:
Please help with a compilation error of the latest version of bugfix-2.0.x motherboard SKR E3 MINI 1.2

In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfigPre.h:54:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:
Marlin\src\HAL\HAL_STM32F1\../../inc/../core/drivers.h:71:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin\src\HAL\HAL_STM32F1\../../inc/../core/drivers.h:106:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/../../Configuration_adv.h:2025:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(Z4)
       ^~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfigPre.h:54:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\HAL_STM32F1\../../inc/../core/drivers.h:71:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin\src\HAL\HAL_STM32F1\../../inc/../core/drivers.h:106:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \

 ^~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/../../Configuration_adv.h:2025:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(Z4)

       ^~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfigPre.h:54:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_STM32F1\Servo.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/../core/drivers.h:71:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin\src\HAL\HAL_STM32F1\../../inc/../core/drivers.h:106:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/../../Configuration_adv.h:2025:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(Z4)
       ^~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/MarlinConfigPre.h:54:0,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/../core/drivers.h:71:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/../core/drivers.h:106:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~

Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/../../Configuration_adv.h:2025:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(Z4)
       ^~~~~~~~~~~
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1

CRCinAU, 2020-01-19:
Possibly related to #16607

sjasonsmith, 2020-01-19:
@carterdo, what source are you using? Your error messages refer to `Z4`, but that isn't merged into any branches yet. That is all still in an open pull request!
`#if AXIS_IS_TMC(Z4)`
#16277

The same is true for your output from the other report, @CRCinAU

CRCinAU, 2020-01-19:
@sjasonsmith mine is a fresh clone of branch bugfix-2.0.x

To ensure nothing weird was going on, I even `rm -fR ~/.platformio ~/git/Marlin/.pio` as well.

I also tried with ststm32 v5.7.0 as well as v6.0.0 - same error. I couldn't find a workaround...

carterdo, 2020-01-19:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4082923/config.zip)

InsanityAutomation, 2020-01-19:
It seems the example configs got merged before the code that supports them.... Try with the code from this PR https://github.com/MarlinFirmware/Marlin/pull/16277 and it should function as expected. Otherwise the configs will need to get rolled back.

sjasonsmith, 2020-01-19:
You can probably just comment out any mention of `Z4` in your configuration files as a workaround for now.

sjasonsmith, 2020-01-19:
@thinkyhead, this issue will need your attention. The example configs are referencing the Z4 feature which isn't merged into any branches yet, causing problems for everybody pulling examples.

carterdo, 2020-01-19:
Thanks, after commenting on the TMC Z4 Marlin compiled

  //#if AXIS_IS_TMC(Z4)
    //#define Z4_CURRENT      800
    //#define Z4_CURRENT_HOME Z4_CURRENT
    //#define Z4_MICROSTEPS    16
    //#define Z4_RSENSE         0.11
    //#define Z4_CHAIN_POS     -1
  //#endif

jedld, 2020-01-19:
a side effect of having the example configs in a seperate repo :)

CRCinAU, 2020-01-20:
Yeah - this is quite painful. Seem like just about every config file has this added - which Marlin main repo doesn't handle.

Is this a case of reverting a single commit to undo the breakage?

GurenWolf, 2020-01-23:
I am getting the same issue using the Bigtreetech SKR 1.3 board and TMC 2208 stepper drivers drivers.

> Thanks, after commenting on the TMC Z4 , Marlin compiled
> 
> //#if AXIS_IS_TMC(Z4)
> //#define Z4_CURRENT 800
> //#define Z4_CURRENT_HOME Z4_CURRENT
> //#define Z4_MICROSTEPS 16
> //#define Z4_RSENSE 0.11
> //#define Z4_CHAIN_POS -1
> //#endif

After commenting this as well, as @carterdo said, my version compiled nicely.

I am using Visual Studio Code and Platformio. These are also my config files in case they are needed (these files don't have the commented version).

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4104822/Marlin.zip)

adrianmihalko, 2020-01-23:
I uncommented if AXIS_IS_TMC(Z4), but I still get errors. I am using SKR3 mini 1.2 on Ender 3, bugfix 2.0.x release:

```
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfigPre.h:54:0,
                 from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:76:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^

Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:112:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/../../Configuration_adv.h:2090:7: note: in expansion of macro 'AXIS_IS_T
MC'
   #if AXIS_IS_TMC(E6)
       ^~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:76:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:112:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \

                               ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/../../Configuration_adv.h:2097:7: note: in expansion of macro 'AXIS_IS_T
MC'
   #if AXIS_IS_TMC(E7)
       ^~~~~~~~~~~
Compiling .pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/Servo.cpp.o
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfigPre.h:54:0,
                 from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/HAL_STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:76:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^

Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:112:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/../../Configuration_adv.h:2090:7: note: in expansion of macro 'AXIS_IS_T
MC'
   #if AXIS_IS_TMC(E6)
       ^~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:76:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:112:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~

Marlin/src/HAL/HAL_STM32F1/../../inc/../../Configuration_adv.h:2097:7: note: in expansion of macro 'AXIS_IS_T
MC'
   #if AXIS_IS_TMC(E7)
       ^~~~~~~~~~~
Compiling .pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp.o
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfigPre.h:54:0,
                 from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/HAL_STM32F1/Servo.cpp:26:
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:76:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:112:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/../../Configuration_adv.h:2090:7: note: in expansion of macro 'AXIS_IS_T

MC'
   #if AXIS_IS_TMC(E6)
       ^~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/../core/drivers.h:76:51: error: missing binary operator before token "("
[... 19 lines cut ...]
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/../../Configuration_adv.h:2090:7: note: in expansion of macro 'A
XIS_IS_TMC'
   #if AXIS_IS_TMC(E6)
       ^~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/../core/drivers.h:76:51: error: missing binary operator before t
oken "("

 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/../core/drivers.h:112:31: note: in expansion of macro 'AXIS_DRIV
ER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) \
                               ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/../../Configuration_adv.h:2097:7: note: in expansion of macro 'A
XIS_IS_TMC'
   #if AXIS_IS_TMC(E7)
       ^~~~~~~~~~~
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/HAL.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/Servo.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/HAL_SPI.cpp.o] Error 1
========================== [FAILED] Took 6.86 seconds ==========================
```

adrianmihalko, 2020-01-23:
After commenting:

  ```
//#if AXIS_IS_TMC(Z4)
  //  #define Z4_CURRENT      800
  //  #define Z4_CURRENT_HOME Z4_CURRENT
  //  #define Z4_MICROSTEPS    16
  //  #define Z4_RSENSE         0.11
  //  #define Z4_CHAIN_POS     -1
  //#endif

  //#if AXIS_IS_TMC(E6)
  //  #define E6_CURRENT      800
  //  #define E6_MICROSTEPS    16
  //  #define E6_RSENSE         0.11
  //  #define E6_CHAIN_POS     -1
  //#endif

  //#if AXIS_IS_TMC(E7)
  //  #define E7_CURRENT      800
  //  #define E7_MICROSTEPS    16
  //  #define E7_RSENSE         0.11
  //  #define E7_CHAIN_POS     -1
  //#endif
```

Compiles, but with default configuration (just enabled bltouch) for this motherboard Auto Home is not working. Separate X, Y, Z homing is working, but If I do G28 printer goes to Ymin, Xmin then Z goes down and nozzle crashes to bed. Weird, because it should go to the center of the bed and then down.

[https://www.youtube.com/watch?v=WdZXVAV5qKI](https://www.youtube.com/watch?v=WdZXVAV5qKI)

carterdo, 2020-01-25:
check if you have uncommented
#define Z_SAFE_HOMING

boelle, 2020-02-02:
@carterdo with latest bigfix 2.0.x do you still have this issue?

CRCinAU, 2020-02-04:
I believe this is fixed now - I can't replicate it anymore with updated configs / bugfix-2.0.x.

It should probably be closed.

boelle, 2020-02-04:
@carterdo do you agree?

PM3Delta, 2020-02-09:
Works for me. I just compiled for the SKR E3 Mini v1.2 today using bugfix-2.0.x and the STM32RC6_bigtree Environment

closed 2020-02-10 by thinkyhead