MarlinFirmware/Marlin #16612: [BUG] the trigorila 14 pins it's wrong , the Y endstop is wront go it's never trigered.

Nestor75, 2020-01-19:
### Bug Description

 the trigorila 14 pins it's wrong , the Y endstop is wront go it's never trigered. 
I had to use an ald trigorilla 14 pins file in order to make it work with my anycubic 4 max pro
this is the one I had to use istead the one in Marlin 2.0.1
[pins_TRIGORILLA_14.txt](https://github.com/MarlinFirmware/Marlin/files/4082873/pins_TRIGORILLA_14.txt)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

install in any anycubic printer and home y axis...

**Expected behavior:**stop the Y axis whtn hits the end stopd

**Actual behavior:**it never triggers the endstop

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-01-23:
what version of marlin do you use?

is the pins file also wrong in bugfix 2.0.x ?

Nestor75, 2020-01-23:
I used the latest stable release 2.0.1

boelle, 2020-01-23:
try with bugfix also, 2.0.1 is older than bugfix

Nestor75, 2020-01-24:
ok, I will do it next time I update the FW. Many thannks for your help :)

thinkyhead, 2020-01-27:
How does the Trigorilla 1.3 board definition look? Similar issues?

I'll add this to `pins_TRIGORILLA_14.h`:
```cpp
//
// Limit Switches
//
#define X_MAX_PIN          43
#define Y_MIN_PIN          19
```

boelle, 2020-01-28:
@Nestor75 should have been fixed now, will close this one

we can always reopen it

closed 2020-01-28 by boelle

Nestor75, 2020-01-29:
pins in Marlin 2.0.2 for trigorilla 1.4 are fine

sjasonsmith, 2020-02-01:
@Nestor75, this change ended up breaking Trigorilla users on other printers. It looks like your printer uses a daughterboard that changes pin numbering compared to other Anycubic printers.

@thinkyhead updated the pins file in the bugfix-2.0.x branch to make the new assignments optional. You will need to enable the ANYCUBIC_4_MAX_PRO_ENDSTOPS option for your printer. You can either uncomment it in the pins file or add it to one of your configuration files.
`#define ANYCUBIC_4_MAX_PRO_ENDSTOPS`

If you have a configuration working well for this printer, I suggest you head over to the configurations repository and submit it as an example!
https://github.com/MarlinFirmware/Configurations

Nestor75, 2020-02-01:
Hi @sjasonsmith 
The board, in theory it is exactly the same trigorilla than the rest of anycubic printers. The Endstop pins in the latest Marlin 2.0.2 works fine but the fan controler PIN, the runout filament PIN  and the led Pin for this specific 4max pro printer do not. 

Anyway, I used and old pins file pins_TRIGORILLA_14.h , I modified the Marlin 2.0.2 code for using the TFT touch screen thanks to @poket-jony guithub and it is workig great but the a part from the configuration files you have to change many other files to make it works. 

I published the code at https://www.thingiverse.com/thing:4104593 sorry for not using github, I am not familiar with code management systems. 

I will try to create a github for that. 
Many thanks all of you for your support.