MarlinFirmware/Marlin #16614: [BUG] Manual bed leveling stops after homing

Vladimir-VS, 2020-01-19:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-01-20:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Vladimir-VS, 2020-01-20:
Added to the original comment.

boelle, 2020-01-21:
@Vladimir-VS please retest with bugfix 2.0.x from here https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

Vladimir-VS, 2020-01-22:
Thank you. Will do at the nearest possibility.

boelle, 2020-01-23:
closing this since its a dublicate of #13591

we only need one open for the same issue

closed 2020-01-23 by boelle