MarlinFirmware/Marlin #16616: Marlin manual mesh bed level not working as expected

DankyDuck82, 2020-01-19:
Hi

I am new to this site, and in general new to 3D printing.
I acquired two Ender 3 pro about a month ago and just updated my board to an SKR Mini E3 v1.2, and because of this I want to use the manual mesh leveling function as my bed is very warped and basically makes it impossible to do larger prints.

Instead of explaining what the problem is, I will take you through my process and frustration that cost me 9 hours so far with the same result (yes, i am definitely doing something wrong.... But what?) 

- Mounted SKR Mini E3 v1.2
- Downloaded Marlin 2.0.1 and made the changes needed to use mesh bed level -> Teaching Tech -> https://www.youtube.com/watch?v=vcxM7-VK44k&t=84s (Configuration.h attached, in a slaughtered version because I can only upload 19kb files)
- Flash firmware using SD card, everything looks like expected.
- I now tighten all the springs to the top, and release them clockwise 1 whole round
- Heat bed to 60
- Manually lift the Z and center it on bed
- Lower Z until it tightens on a piece a paper
- Adjust the Z end switch to fit the newly adjusted level.
- Autohome with 60 heated bed
- Select level bed corners within the new LCD functions
- Do all the four corners with a piece of paper until I am satisfied with the result.
- Now I do the Manual mesh bed level function with all 9 points on the bed. Very happy with the result
- I store the data.
- Now for a test print, while adjusting the Bed Z on the fly -> Test print from CHEP > https://www.thingiverse.com/thing:3235018
- Test print needed some adjustments but only using the LCD bed adjustment for this.
- Store data. Ready for print
- The printer heats up, I am excited (or really not, because of this maybe the 20th time, and I already know the result)
- AND YES! 21st time also failed, the nozzle is grinding my glass plate, adjusting the Bed Z with my LCD goes from -0.327 to +0.200 and it is still hitting the glass bed.

I have tried reflashing the firmware ultiple times, I have checked I have activated

#define RESTORE_LEVELING_AFTER_G28,

and added GCODE to my CURA settings

G28 ;Home
M420 S1 ;enable the mesh you previously stored in eeprom.

So I am completely lost, Manual Mesh bed level should have been my savior, but its far from it....!!

Any help would be appreciated (Just if you reached this part, reading everything, that is appreciated  )

DankyDuck82, 2020-01-19:
Here is my configuration file. 
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4083228/Configuration.txt)

boelle, 2020-01-19:
not an expert, but have you tried with bugfix 2.0.x ? that one is newer than 2.0.1

DankyDuck82, 2020-01-19:
> not an expert, but have you tried with bugfix 2.0.x ? that one is newer than 2.0.1

No, I have not, I got confused about what version to choose, but maybe I should try it.

reloxx13, 2020-01-22:
> while adjusting the Bed Z on the fly

if u do this, ull need to redo manual mesh leveling afterwards.
ur home will be adjusted and afterwards ur mesh wil be wrong now.

im not sure if m851 zxx/-xx will work with manual mesh, but usually this value is for finetuning after mesh leveling or use babystepping feature.

boelle, 2020-01-26:
@DankyDuck82 did you try bugfix 2.0.x ?

DankyDuck82, 2020-01-26:
Yes I tried it and the situation was the same. I found another solution that required me not making any test prints before doing other prints and that seems to have solved what ever was causing it. Now I am just babystepping the first layer.

boelle, 2020-02-02:
> @DankyDuck82 did you try bugfix 2.0.x ?

care to share? might help the code people

YS-Palumbo, 2020-05-29:
Unfortunatly i got the exact same problem.
I also got the SKR Mini E3 V1.2.

MESH_BED_LEVELING and RESTORE_LEVELING_AFTER_G28 are enabled.
M420 S1 in the Startcode also doesn't work.

Is there any fix ?

boelle, 2020-06-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-21 by boelle