MarlinFirmware/Marlin #16619: [BUG] Endstops swapped in delta (SKR 1.3 / TMC2209 / Anycubic Predator)

StefanHamminga, 2020-01-20:
### Bug Description

My Anycubic Predator uses a BTT SKR 1.3 mainboard with TMC2209 v1.2 drivers in UART mode. I've also added the bridge jumpers to be able to use the TMCs as `{X,Y,Z}-` endstops (with the intent to do sensorless probing). The OEM endstops are connected to the `{X,Y,Z}+` headers of the board.

According to the board [schematic](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-SCH.pdf) this should work fine.

When homing in Marlin (upwards, towards the `+` stops) the carriages crash into the frame and eventually Marlin halts. Using the lights and multimeter I can confirm the homing switches work fine and they are properly configured as non-inverting (active high). Moving the axes manually in OctoPrint and checking with `M119` shows that moving an axis to trigger an endstop actually shows as the `-` endstop being triggered in Marlin (and subsequently ignored). This goes for all axes, individually or together.
I've tried removing the jumpers for sensorless negative endstops, dis/enabling most software features, but for some reason the issue stays the same.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4085904/config.zip)

### Steps to Reproduce

See above.

**Expected behavior:** Homing works, positive endstops get registered

**Actual behavior:** Carriages crash into frame, negative endstops show state `TRIGGERED`.

ellensp, 2020-01-20:
pins file contains code 
`#if X_HOME_DIR == -1 || !X_STALL_SENSITIVITY
  #define X_MIN_PIN          P1_29   // X_MIN
  #define X_MAX_PIN          P1_28   // X_MAX
#else
  #define X_MIN_PIN          P1_28   // X_MAX
  #define X_MAX_PIN          P1_29   // X_MIN
#endif`
so if you home to max you still use the min pins.

StefanHamminga, 2020-01-20:
Thanks for pointing that out! I disabled the logic and homing worked.
The attached patch changes the logic to not swap pins if ..._MAX is defined, preserving the original idea for the logic, but fixing this issue.

[0001-Do-not-invert-pins-if-using-dedicated-MAX-endstop.zip](https://github.com/MarlinFirmware/Marlin/files/4087110/0001-Do-not-invert-pins-if-using-dedicated-MAX-endstop.zip)

boelle, 2020-01-23:
@StefanHamminga so this one is solved?

StefanHamminga, 2020-01-23:
Well, the unexpected behaviour is still present in Marlin, I wouldn't call manually editing code 'fixed', but the source is found. Despite the good intentions, swapping endstops without explicit notification or user interaction is a recipe for disaster imho.

ellensp, 2020-01-23:
yea, ill turn your patch into a PR and see if the marlin admins think its a good idea to implement or not..

InsanityAutomation, 2020-01-23:
Current convention is using the #_STOP_PIN for a singular endstop either direction. If the pins file isn't set up for it, compare to ramps.

ellensp, 2020-01-26:
Equivalent code has been added to bugfix to allow for USE_XMAX_PLUG

StefanHamminga, 2020-01-26:
Thanks all for fixing this!

closed 2020-01-26 by StefanHamminga