MarlinFirmware/Marlin #16623: BLTouch/3D touch help with mks base v1.4 board

bentech4u, 2020-01-20:
After some software changes the behavior changed

M280 P0 S10 : pin goes down : m119 shows open

M280 P0 S90: pin goes in : m119 shows triggered

, but when i try to home all, after touching bed also z axis is moving down and it is stopping only when the 3dtouch goes into alarm state

[https://vimeo.com/user32803137/review/385998054/629786ec31](url)

boelle, 2020-01-20:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

bentech4u, 2020-01-21:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4088955/Marlin.zip)
here i am attaching configurations

boelle, 2020-01-30:
the config does not match current bugfix 2.0.x

where did you download marlin from?

Vertabreak, 2020-01-30:
the following does not work with 3dtouch probes.
#define BLTOUCH_FORCE_SW_MODE

yet-another-average-joe, 2020-02-03:
@bentech4u It seems your probe is mounted a bit too low. (hard to say on the video) I have also been using a 3DTouch (the black one), and it sometines drops the pin 2 times insteed of one. It behaved weirdly (and had poor repeatability)
Also, there could be some EMI with the heaters ; maybe you could try PROBING_HEATERS_OFF

Vertabreak, 2020-02-03:
EMI is defiantly a issue on all the GT2560 boards there are several options that can help deal with it.

boelle, 2020-02-08:
since we have not gotten any word from the OP we must either assume he solved his problem or that he have given up

will close this one and we can reopen if OP returns

closed 2020-02-08 by boelle