MarlinFirmware/Marlin #16624: [BUG] filament unload during print

EugeneSkrebnev, 2020-01-20:
First of all i am sorry if i am doing smth wrong with configuration files.

while printing this gcode [Bed_Level_Test.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/4087386/Bed_Level_Test.gcode.txt) in left bottom corner 
 in the middle of infill suddenly print pause and filament unload

### My Configurations
UBL enabled (i try to clear epprom and recalibrate, nothing changes)
sd card 4gb i try to format it - nothing changes - always the same place unload... not random unload.
full code 2.0.1 branch [dropbox link to full code](https://www.dropbox.com/s/xeqcghr1o7jy150/SK-GO-skr13.zip?dl=0)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4087384/Configuration.zip)

### Steps to Reproduce
-shutdown printer
-turn on
-print from media (*sd card)
-gcode [Bed_Level_Test.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/4087386/Bed_Level_Test.gcode.txt)
-wait till left bottom corner infill starts - unexpected filament unload in the middle of infill.
-I'm so CONFUSED that I created this ticket

**Expected behavior:** 
no filament unload was expected

**Actual behavior:** [What actually happens]
unexpected filament unload

Vertabreak, 2020-01-21:
hate to ask but i have to the filament was inside the sensor when it triggered yes?

EugeneSkrebnev, 2020-01-21:
Good question but not so easy)... for now i will disable sensor completely in config and try one more time... but 3 times printing it fails in exactly same place of gcode

Vertabreak, 2020-01-21:
have you tried bugfix branch it is newer then stable and its possible the issue had been patched.

EugeneSkrebnev, 2020-01-21:
it works after fixing insets... close for now, strange but ok for me for now.

thinkyhead, 2020-01-21:
I was going to suggest that if you know that you have enough filament, disable the filament sensor. There might be something tugging on it to cause it to lose connection every so often, so best to be safe.

closed 2020-01-21 by EugeneSkrebnev

EugeneSkrebnev, 2020-01-21:
> I was going to suggest that if you know that you have enough filament, disable the filament sensor. There might be something tugging on it to cause it to lose connection every so often, so best to be safe.

that doesnt look like true because i run this gcode 3 times and 3 times it fails in exactly same position... looks like it was software case... but i mot sure,

big thanks for everybody

thinkyhead, 2020-01-21:
Marlin would only initiate a filament change if it saw the filament sensor pin state change. It does not initiate filament change under other circumstances. So it is most likely a hardware issue.

Liger0, 2020-02-02:
It's the first time I have this same issue in 2.5 years, and it only happened after updating to the bugfix of the 21 January.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4144394/Marlin.zip)

thinkyhead, 2020-02-07:
> it only happened after updating 

_**Post hoc ergo propter hoc?**_

How reproducible is the issue? Again, Marlin doesn't unload filament unless the filament sensor state changes.

To mitigate false readings you can increase `FILAMENT_RUNOUT_THRESHOLD` to a higher value (default is 5). And you can also enable `FILAMENT_RUNOUT_SENSOR_DEBUG` to get extra logging that might provide a clue.