MarlinFirmware/Marlin #16625: [BUG] Non-symmetrical behavior of the dondolo tool change

Vladimir-VS, 2020-01-20:
### Bug Description

On the Tevo Tarantula with 2 Nozzles on the dondolo mount
At the change of a tool prior to homing the hotend is **not** raised by the TOOLCHANGE_ZRAISE value but after the change is lowered down by the TOOLCHANGE_ZRAISE value thus causing repetitive lowering of the nozzle.

The issue is cased by the difference in "no_move" validation in the tool_change.cpp module.
If changing the line 936
        if (!no_move) fast_line_to_current(Z_AXIS);
to
        if (!no_move) ;
	fast_line_to_current(Z_AXIS);
the nozzle is properly hopped even prior to homing.

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4087427/Configuration.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

GMagician, 2020-01-20:
Sure are you using new marlin? This should be fixed now. When not home Marlin doesn't move Z

boelle, 2020-01-21:
@Vladimir-VS please retest with bugfix 2.0.x from here https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

Vladimir-VS, 2020-01-22:
Indeed, Zaro, this should fix the issue. Please, close my ticket if you are ok.

GMagician, 2020-01-22:
I can't close it...you have to do that with button on the bottom side of the page

closed 2020-01-23 by boelle