MarlinFirmware/Marlin #16626: [BUG] PID work not ok

userosos, 2020-01-20:
Marlin after PID autotune not set to target Temperature extruder. See the pic please!
Log from repetier-server. Mks gen L.
Heater for 12V. power supply is 24V. #define BANG_MAX 70
[Marlin-2.0.1.zip](https://github.com/MarlinFirmware/Marlin/files/4087529/Marlin-2.0.1.zip)
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4087530/Marlin-bugfix-2.0.x.zip)
Marlin 2.0.1 or Bugfix2.0.x
[image]

userosos, 2020-01-21:
I think thah it will be problem with connector from termistor of the hotend.

boelle, 2020-01-23:
closing as it was a connector issue

closed 2020-01-23 by boelle