MarlinFirmware/Marlin #16627: [BUG] SKR V1.3 + BLTOUCH prevent Z-axis movement

jaisoh, 2020-01-20:
### Bug Description

Using Marlin 2.0.1

When enabling BLTouch from configuration the Z-axis stops working. When trying to move Z-axis I only hear loud click and it seems that the SKR V1.3 is stuck and after a while halts. When BLTouch is not enabled all the axis move correctly.

PS. I think I have clone BLTouch but it was working fine with marlin 1.1.9 and Ender 3 original board until one stepper driver died from the board and I purchased SKR V1.3 so I can use Trinamic drivers.

### My Configurations

Will upload files later as I have to revert back to non-working configuration.
I took Ender 3 example from Marlin 2.0.1 and uncommented BLTouch relevant lines.

### Steps to Reproduce

1. Uncommented #Define BLTOUCH from Configuration.h 
2. Build the FW.
3. Uploaded FW to the printer.

**Expected behavior:** 
- Z-axis should move when BLTouch is enabled and not halt the CPU.

**Actual behavior:**
-  Z-axis stepper clicks and does not move. LCD UI does respond also when Z-axis movement is triggered.

boelle, 2020-01-23:
try with bugfix 2.0.x instead, its newer than 2.0.1

houseofbugs, 2020-01-23:
Does it work without BLTOUCH uncommented?

jaisoh, 2020-01-23:
Been debugging this for few days and tried the bugfix 2.0.x yesterday and got it working fine. 

What I found out was that when the Hybrid option of TMC is turned on and printer is trying to do fast movements the axis does not move. I tuned the speed thresholds for hybrid mode higher to prevent the TMCs change mode to spreadcycle and axis movement started working.

Probably something wrong with my TMC2130 setup. Need to take a closer look at work tomorrow are the PCBs ok, I'm suspecting CFG4/CFG5 jumpers might cause this issue.

So I think it has nothing to do with BLTouch and therefore I'm closing the issue. Sorry for the inconvenience.

closed 2020-01-23 by jaisoh