MarlinFirmware/Marlin #16628: [BUG] TOUCH_UI_FTDI_EVE doesnt work.

TB1405, 2020-01-20:
### Bug Description
The TOUCH_UI_FTDI_EVE does not work on marlin 2.0.1 or 2.0.x
i tried on Ramps, Arduino, SKR Pro, Fysetc S6 and SKR 1.3 
the compiling issue https://github.com/MarlinFirmware/Marlin/issues/16534 is sloved. 
but now only the display back light is on. i tried a lot different configurations but the result was every time the same. 
Green screen while booting  and after this backlight is on and display black. 
Only if disconnect the MISO pin, on the display is the notification `Release to begin screen calibration`
[image]
[image]

with the marlin version from fysetc the display works pefect. 
https://github.com/FYSETC/Marlin-2.0.x-FYSETC/tree/S6/BUG-FIX-CLCD
[image]

i also copied the folder `ftdi_eve_touch_ui ` from fysetcs marlin in the marlin 2.0.1 branch after this the display works also but only with s6 board and SKR pro. 
copy the `ftdi_eve_touch_ui ` folder to the actual bugfix version does not work because there comes a lot of errors.

### My Configurations

here are my configs that i actual work on my printer. (TOUCH_ui Folder is from fysetc) 
[Config_SKRPRO_MMU2_FTDI2.0.1WithFysetcTouchUI.zip](https://github.com/MarlinFirmware/Marlin/files/4087645/Config_SKRPRO_MMU2_FTDI2.0.1WithFysetcTouchUI.zip)

### Steps to Reproduce

1. download marlin 2.0.1 
2. use my config and connect a ftdi display to the board 
3. compile it with original `ftdi_eve_touch_ui ` folder from the marlin branch **Expected behavior:** [Display doesnt work]
3. compile it again with fysetc `ftdi_eve_touch_ui ` from the marlin branch **Expected behavior:** [Display does work]

#### Additional Information
I also noticed that the #defines from the cofiguration.adv are not taken over.
As an example you select //#define CLCD_USE_SOFT_SPI in config.adv but the display still works with the normal SPI pins on EXP2.

but if you change it as follows in the pin_mappings.h

```
#ifdef SKR_PRO_TFT_PINMAP
  #ifndef __MARLIN_FIRMWARE__
    #error This pin mapping requires Marlin.
  #endif

  #define CLCD_SPI_CS PG10 //PD11 
  #define CLCD_MOD_RESET PF11 //PG2 
  
  //#define CLCD_USE_SOFT_SPI
    // #define CLCD_SOFT_SPI_MOSI PG3
    //#define CLCD_SOFT_SPI_MISO PG4
    //#define CLCD_SOFT_SPI_SCLK PA8 // PORTA3 Pin 7             
#endif
```

the display is now  controlled by the defined pins (backlight goes on) but unfortunately not correctly because the display is only switched on but nothing is displayed.  so Software_SPI isnt work correctly
the same problem even if the CS and PD pin is only defined in config.adv the display is not controlled at all. 
also this must happen in pin_mappings.h as mentioned above. 

#define CLCD_SPI_CS PG10 //PD11 
#define CLCD_MOD_RESET PF11 //PG2 
`
next problem is now 
if we use the LCD on the hardware spi bus we cant connect a sd card module. 
as soon as i have connected the sd card module to the spi bus of the display the display shows only strange things 
is it possible to use sd card modul with software spi?

I hope this is written understandably.

thinkyhead, 2020-01-22:
Someone will have to finish porting support for this display into `bugfix-2.0.x` using the FYSETC fork as a reference.

TB1405, 2020-01-22:
@thinkyhead maybe you have any idea because the issue with sd card ? 
and one more question will this display be supported in the future if for example new submenus are created. As far as I know these would have to be created in the FTDI Eve program.

marciot, 2020-02-24:
I don't know whether this thread is still active, but I suggest putting a resistor on the MISO line, as someone else had run into problems without one. If in fact there is a software fix, I would find it to be interesting and worth additional investigation.

marciot, 2020-02-24:
> My SD Card modul is the problem i connect the sd card modul from the smart discount controller and it works.

Interesting. I do not recommend sharing the SPI bus between the FTDI display and the SD card. If you need to use an SD card, connect it to the SPI pins and run the FTDI display off some other pins using the software SPI option.

> one more question will this display be supported in the future if for example new submenus are created. As far as I know these would have to be created in the FTDI Eve program.

No, the display code was developed at LulzBot (I wrote it). We could not use FTDI's closed-source tools so we wrote our own code from scratch. Some menus might get added in the future if anyone pays me to do it. Like right now I doing some work for CocoaPress :)

thisiskeithb, 2020-02-24:
I’m watching it since I have an FT810 that I couldn’t get working with Marlin’s version on an SKR 1.3 (never tried Fysetc’s/don’t have an S6 board).

sl1pkn07, 2020-04-14:
my touch_eve is arrived (Fysetc 5" (https://es.aliexpress.com/item/4000628057351.html)). i want to test it in a SKR E3 DIP and Ramps+ReARM

~~~
#define TOUCH_UI_FTDI_EVE
#define LCD_HAOYU_FT810CB
#define CR10_TFT_PINMAP
~~~

my first test i have hit the error https://github.com/MarlinFirmware/Marlin/issues/16534#issuecomment-573382828. add

~~~patch
diff --git a/Marlin/src/HAL/STM32F1/HAL.h b/Marlin/src/HAL/STM32F1/HAL.h
index ff42beb92..48c9e8605 100644
--- a/Marlin/src/HAL/STM32F1/HAL.h
+++ b/Marlin/src/HAL/STM32F1/HAL.h
@@ -47,6 +47,9 @@
   #include "msc_sd.h"
 #endif
 
+
+#define pgm_read_ptr_near pgm_read_ptr
+
 // ------------------------
 // Defines
 // ------------------------
~~~

 in src/HAL/STM32F1/HAL.h seems silence the error

now fail in:

~~~
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/../../pin_mappings.h:46:31: error: 'LCD_PINS_ENABLE' was not declared in this scope
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/../../pin_mappings.h:47:31: error: 'LCD_PINS_RS' was not declared in this scope
~~~

add

~~~patch
diff --git a/Marlin/src/pins/stm32f1/pins_BTT_SKR_E3_DIP.h b/Marlin/src/pins/stm32f1/pins_BTT_SKR_E3_DIP.h
index fef8eeb61..f39667fc3 100644
--- a/Marlin/src/pins/stm32f1/pins_BTT_SKR_E3_DIP.h
+++ b/Marlin/src/pins/stm32f1/pins_BTT_SKR_E3_DIP.h
@@ -220,6 +220,15 @@
 
 #endif // HAS_SPI_LCD
 
+#if ENABLED(TOUCH_UI_FTDI_EVE)
+
+  #define BEEPER_PIN              EXPA1_10_PIN
+
+  #define LCD_PINS_RS             EXPA1_04_PIN
+  #define LCD_PINS_ENABLE         EXPA1_03_PIN
+  #define LCD_PINS_D4             EXPA1_05_PIN
+#endif // TOUCH_UI_FTDI_EVE
+
 //
 // SD Support
 //
~~~
i think fix, because now fails in:

~~~
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/advanced_settings_menu.cpp:141:8: error: macro "GET_TEXT" passed 4 arguments, but takes just 1
        ))
        ^
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/confirm_start_print_dialog_box.cpp.o
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/confirm_user_request_alert_box.cpp.o
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/default_acceleration_screen.cpp.o
In file included from /home/sl1pkn07/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/WString.h:29:0,
[... 22 lines cut ...]
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                     ^~~~
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/dialog_box_base_class.cpp.o
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/display_tuning_screen.cpp.o
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/endstop_state_screen.cpp.o
In file included from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/../../../../../inc/../core/language.h:382:0,
                 from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/../../../../../inc/MarlinConfig.h:44,
                 from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/../../../ui_api.h:45,
                 from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/../compat.h:31,
                 from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/../config.h:24,
                 from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/advanced_settings_menu.cpp:23:
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/advanced_settings_menu.cpp: In static member function 'static void AdvancedSettingsMenu::onRedraw(draw_mode_t)':
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/../../../../../inc/../core/multi_language.h:77:53: error: 'GET_TEXT' was not declared in this scope
 #define GET_TEXT_F(MSG) (const __FlashStringHelper*)GET_TEXT(MSG)
                                                     ^
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/advanced_settings_menu.cpp:135:48: note: in expansion of macro 'GET_TEXT_F'
       .tag(7) .button( JERK_POS,               GET_TEXT_F(
                                                ^~~~~~~~~~
*** [.pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/about_screen.cpp.o] Error 1
*** [.pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/advanced_settings_menu.cpp.o] Error 1
~~~

anyone know how fix this?

greetings

marciot, 2020-04-14:
@sl1pkn07: Yah, it's a mistake in "src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/advanced_settings_menu.cpp"

The lines starting on line 135 which read:

```
      .tag(7) .button( JERK_POS,               GET_TEXT_F(
        #if DISABLED(CLASSIC_JERK)
          MSG_JUNCTION_DEVIATION
        #else
          JERK_POS
        #endif
       ))
```

Should be:

```
      .tag(7) .button( JERK_POS,               GET_TEXT_F(
        #if DISABLED(CLASSIC_JERK)
          MSG_JUNCTION_DEVIATION
        #else
          MSG_JERK
        #endif
       ))
```

And in "src/lcd/extui/lib/ftdi_eve_touch_ui/screens/about_screen.cpp"

```
  char about_str[
    strlen_P(GET_TEXT(MSG_ABOUT_TOUCH_PANEL_2)) +
    strlen_P(TOOLHEAD_NAME) + 1
  ];
```

Should be:

```
  char about_str[
    strlen_P(GET_TEXT(MSG_ABOUT_TOUCH_PANEL_2)) +
    #ifdef TOOLHEAD_NAME
      strlen_P(TOOLHEAD_NAME) +
    #endfi
    1
  ];
```

I'll submit a patch.

thisiskeithb, 2020-04-14:
I’d be interested if you can get this working on that board because I could never get it working on an SKR 1.3/1.4 or E3 series.

sl1pkn07, 2020-04-14:
Thanks @marciot now build without problem!

now, the true's hour. flash the firmware....

i've edited my post with s litte changes about the pinout when use the cr10_tft_pinmap. looking with your eyes, is ok?

greetings

sl1pkn07, 2020-04-14:
hardware conected and not works :/

anyone have the pinout of this screen? @GerogeFu ?

marciot, 2020-04-14:
@sl1pkn07: Try one of two things:

1) Power on the board with your finger on the display. Hold it there for a few seconds while the board starts up. This enables a failsafe mode which may help (especially if the display is dimmed all the way).
2) If that does not do the trick, enable TOUCH_UI_DEBUG in "src/lcd/extui/lib/ftdi_eve_touch_ui/config.h" and look for errors in the serial console.

sl1pkn07, 2020-04-14:
nope the 1). the 2) IDK how get the serial console

one question. in the S6, is nneed conect to EXP2, rigth?

i fount this:
[image]
[image]

but for example, in the E3 DIP, the connector is flipped
[image]

flip the connector (disassemple in the cord and turn 180º) seems now have the rigth connect . but the screen is still blank

marciot, 2020-04-14:
Without having this board myself, I can' really offer much guidance. There are pin specific configuration options in:

- src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\config.h
- src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\pin_mappings.h

Also, there are board and display specific configurations in:

- src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\basic\boards.h
- src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\basic\resolutions.h

All these things need to line up correctly for the display to work. If anyone wants to donate one of these displays, I can figure out how to make it work, otherwise this is pretty much all the guidance I can offer.

thisiskeithb, 2020-04-14:
@marciot: If I can get you some SKR boards, would you be able to help get this TFT/screen type working in `2.0.x`?

sl1pkn07, 2020-04-14:
i also create a custom adapter with dupont wires

~~~
  /** Fysect Touch EVE 5" display pinout
   *                   _____
   *             MISO | 1 2 | SCK
   *        MOD_RESET | 3 4 | SD_CS
   *           LCD_CS | 5 6   MOSI
   *           SD_DET | 7 8 | RESET
   *              GND | 9 10| 5V
   *                   -----
   *                    EXP1
   */
~~~

from Fysect S6 EXP2 pinout and CR10_TFT_PINMAP logic in pin_mappings.h

not works

i also make a diff betwen https://github.com/FYSETC/Marlin-2.0.x-FYSETC/tree/S6/MASTER and marlin-bugfix-2.0.x and i not see related difference with respect the Touch_EVE

[Diff_fysect_s6_marlin_bugfix-2.0.x.diff.txt](https://github.com/MarlinFirmware/Marlin/files/4477272/Diff_fysect_s6_marlin_bugfix-2.0.x.diff.txt)

maybe need a more power?

greetings

marciot, 2020-04-14:
> @marciot: If I can get you some SKR boards, would you be able to help get this TFT/screen type working in `2.0.x`?

If you were willing to donate an SKR board and a TFT panel of the type you need working, I could look into getting it supported. My guess is that there are parameters that aren't quite right.

marciot, 2020-04-14:
Try disabling SD_SUPPORT. At least for starters. It might be an incompatibility there.

thisiskeithb, 2020-04-14:
I’ll ping BigTreeTech about the boards and get back to you.

I only have one FTDI EVE FT810 at the moment and I’m not even sure if it works since I could never get anything to display on the screen.

marciot, 2020-04-14:
@thisiskeithb: It's better to take things one step at a time. There are some Arduino sketches here that should light up the display:

https://github.com/marciot/drunken-octopus-marlin/tree/master/extras-tools/ftdi-eve-lib-examples

If you can get *that* to work, then the next step is to try getting it to work in Marlin and on the LPC chip.

sl1pkn07, 2020-04-14:
not works for me (logo demo) with Arduino mega (Taurino).

build ok. pins wired. the sketch is upload ok. but zero messages in serial console and no light in the LCD

marciot, 2020-04-14:
Did you change the pin definitions in "logo_demo/src/config.h"?

marciot, 2020-04-14:
Also, what display board are you defining here?

```
//#define LCD_FTDI_VM800B35A        // FTDI 3.5" 320x240 with FT800
//#define LCD_4DSYSTEMS_4DLCD_FT843 // 4D Systems 4.3" 480x272
//#define LCD_HAOYU_FT800CB         // Haoyu with 4.3" or 5" 480x272
//#define LCD_HAOYU_FT810CB         // Haoyu with 5" 800x480
//#define LCD_ALEPHOBJECTS_CLCD_UI  // Aleph Objects Color LCD User Interface
```

sl1pkn07, 2020-04-14:
i leave it untouched (is set `LCD_HAOYU_FT810CB`, like i set in the marlin)

sl1pkn07, 2020-04-15:
tested with software spi (`#define CLCD_USE_SOFT_SPI` pins 11,12,13), hardware spi (`//#define CLCD_USE_SOFT_SPI` pins 50,51,52, include ICSP pins). no works

only do a little thing when feed the lcd with external power (dc-dc 24v->5v). the speaker sound with one little pulse when plug the power

sl1pkn07, 2020-04-15:
Works with @rudolphridel FT800-FT813 examples https://github.com/RudolphRiedel/FT800-FT813.git (the only not works is the touch)

RudolphRiedel, 2020-04-15:
The difference is that the HY50HD has a crystal populated.
Try commenting out the "#define EVE_HAS_CRYSTAL" in EVE_config.h für the EVE_FT810CB_HY50HD.
That may or may not be the issue then with Marlin as well.

Which reminds me that I wanted to play some more with Marlin to make my Fysetc Cheetah connect to an EVE based display.
The Cheetah has a hardware SPI on the display connector.

marciot, 2020-04-15:
> The difference is that the HY50HD has a crystal populated.

I was thinking about that too. When trying unsupported displays, be sure to try LCD_ALEPHOBJECTS_CLCD_UI as well as LCD_HAOYU_FT810CB. The former has a crystal while the latter does not. So trying both options might allow other boards to work. The "boards.h" file defines this and other attributes.

sl1pkn07, 2020-04-15:
Hi

tested you changes and works except the touch and the same sound in the speaker

RudolphRiedel, 2020-04-15:
Has Fysetc released a schematic for the display?
I briefly thought about buying one of these but it is only resistive touch and only has a FT810.

marciot, 2020-04-15:
If anyone wants to buy me one of those Fysetc TFT81050 for me, I'll make it work :)

sl1pkn07, 2020-04-15:
oh!

[image]

the touch still not works /Nintendo DS stylus eiter xd)

@thisiskeithb you know if the SPI pins in the E3 DIP (display port) is hardware spi?

sl1pkn07, 2020-04-15:
with hardware SPI

~~~
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/base_numeric_adjustment_screen.cpp.o
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/spi.cpp: In function 'void FTDI::SPI::spi_read_bulk(void*, uint16_t)':
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/spi.cpp:101:28: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, uint16_t&)'
       ::SPI.transfer(p, len);
                            ^
In file included from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/spi.h:26:0,
                 from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/ftdi_basic.h:37,
                 from Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/spi.cpp:23:
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/base_screen.cpp.o
Marlin/src/HAL/STM32F1/SPI.h:268:11: note: candidate: uint8_t SPIClass::transfer(uint8_t) const
   uint8_t transfer(uint8_t data) const;
           ^~~~~~~~
Marlin/src/HAL/STM32F1/SPI.h:268:11: note:   candidate expects 1 argument, 2 provided
~~~

RudolphRiedel, 2020-04-15:
I tried to reconstruct the display paramets for TOUCH_UI_800x480 from Marlin.

resolutions.h

#define COMPUTE_REGS_FROM_DATASHEET \
    constexpr uint16_t Hoffset              = thfp + thb - 1; \
    constexpr uint16_t Hcycle               = th; \
    constexpr uint16_t Hsync0               = thfp - 1 ; \
    constexpr uint16_t Hsync1               = thfp + thpw - 1; \
    constexpr uint16_t Voffset              = tvfp + tvb - 1; \
    constexpr uint16_t Vcycle               = tv; \
    constexpr uint16_t Vsync0               = tvfp - 1; \
    constexpr uint16_t Vsync1               = tvfp + tvpw - 1; \
    static_assert(thfp + thb + Hsize == th, "Mismatch in display th"); \
    static_assert(tvfp + tvb + Vsize == tv, "Mismatch in display tv");

#elif defined(TOUCH_UI_800x480)
  namespace FTDI {
    constexpr uint8_t  Pclk                 =    3;
    constexpr uint8_t  Pclkpol              =    1;
    constexpr uint16_t Hsize                =  800;
    constexpr uint16_t Vsize                =  480;

    constexpr uint16_t th                   = 1056; // One horizontal line
    constexpr uint16_t thfp                 =  210; // HS Front porch
    constexpr uint16_t thb                  =   46; // HS Back porch (blanking)
    constexpr uint16_t thpw                 =   23; // HS pulse width

    constexpr uint16_t tv                   =  525; // Vertical period time
    constexpr uint16_t tvfp                 =   22; // VS Front porch
    constexpr uint16_t tvb                  =   23; // VS Back porch (blanking)
    constexpr uint16_t tvpw                 =   10; // VS pulse width

Is this really working for any 800x480 display? These are the most exotic settings I have seen so far.

VSYNC0 = 21
VSYNC1 = 32
VOFFSET = 44
VCYLCE = 525
HYNC0 = 209
HYSNC1 = 232
HOFFSET = 255
HCYCLE = 1056

This might be working better, this is a lot closer to the 20+ displays I support with my library:

#elif defined(TOUCH_UI_800x480)
  namespace FTDI {
    constexpr uint8_t Pclk                 =    2;
    constexpr uint8_t Pclkpol              =    1;
    constexpr uint16_t Hsize               =  800;
    constexpr uint16_t Vsize               =  480;
    constexpr uint16_t Vsync0              =    0;
    constexpr uint16_t Vsync1              =    2;
    constexpr uint16_t Voffset             =   32;
    constexpr uint16_t Vcycle              =  525;
    constexpr uint16_t Hsync0              =    0;
    constexpr uint16_t Hsync1              =   48;
    constexpr uint16_t Hoffset             =   88;
    constexpr uint16_t Hcycle              =  928;

    constexpr uint32_t default_transform_a  =  0x0000D8B9;
    constexpr uint32_t default_transform_b  =  0x00000124;
    constexpr uint32_t default_transform_c  =  0xFFE23926;
    constexpr uint32_t default_transform_d  =  0xFFFFFF51;
    constexpr uint32_t default_transform_e  =  0xFFFF7E4F;
    constexpr uint32_t default_transform_f  =  0x01F0AF70;
  }

sl1pkn07, 2020-04-15:
Hi again. now the touchscreen works. seems need calibrate first (keep press with the finger at boot and the follow the instructions in the screen (push some dots around the display))

seems the UI is unusable. and when press the butons do nothing. only sound click in the speaker

i have made this changes in the main marlin if anyone can check it if is ok (based on bugfix-2.0.x)

~~~patch
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index b0a0f8ebb..c2c521326 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -104,13 +104,13 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT -1
 
 /**
  * Select a secondary serial port on the board to use for communication with the host.
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-//#define SERIAL_PORT_2 -1
+#define SERIAL_PORT_2 2
 
 /**
  * This setting determines the communication speed of the printer.
@@ -128,7 +128,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_BTT_SKR_E3_DIP
 #endif
 
 // Name displayed in the LCD "Ready" message and Info menu
@@ -2077,7 +2077,7 @@
 // Touch UI for FTDI EVE (FT800/FT810) displays
 // See Configuration_adv.h for all configuration options.
 //
-//#define TOUCH_UI_FTDI_EVE
+#define TOUCH_UI_FTDI_EVE
 
 //
 // Third-party or vendor-customized controller interfaces.
diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
index fbfe63036..50d4fb325 100644
--- a/Marlin/Configuration_adv.h
+++ b/Marlin/Configuration_adv.h
@@ -1343,18 +1343,19 @@
   //#define LCD_4DSYSTEMS_4DLCD_FT843 // 4D Systems 4.3" (480x272)
   //#define LCD_HAOYU_FT800CB         // Haoyu with 4.3" or 5" (480x272)
   //#define LCD_HAOYU_FT810CB         // Haoyu with 5" (800x480)
-  //#define LCD_ALEPHOBJECTS_CLCD_UI  // Aleph Objects Color LCD UI
+  #define LCD_ALEPHOBJECTS_CLCD_UI    // Aleph Objects Color LCD UI
 
   // Correct the resolution if not using the stock TFT panel.
   //#define TOUCH_UI_320x240
   //#define TOUCH_UI_480x272
-  //#define TOUCH_UI_800x480
+  #define TOUCH_UI_800x480
 
   // Mappings for boards with a standard RepRapDiscount Display connector
   //#define AO_EXP1_PINMAP    // AlephObjects CLCD UI EXP1 mapping
   //#define AO_EXP2_PINMAP    // AlephObjects CLCD UI EXP2 mapping
   //#define CR10_TFT_PINMAP   // Rudolph Riedel's CR10 pin mapping
   //#define S6_TFT_PINMAP     // FYSETC S6 pin mapping
+  #define SKR_E3_PINMAP       // BIGTREETECH SKR E3 board series pin mapping
 
   //#define OTHER_PIN_LAYOUT  // Define pins manually below
[... 106 lines cut ...]
-    #error "Only CR10_STOCKDISPLAY, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_E3_DIP."
+    #error "Only CR10_STOCKDISPLAY, TOUCH_UI_FTDI_EVE, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_E3_DIP."
   #endif
 
-#endif // HAS_SPI_LCD
+#endif // HAS_SPI_LCD || ENABLED(TOUCH_UI_FTDI_EVE)
 
 //
 // SD Support
diff --git a/platformio.ini b/platformio.ini
index 38e3065f2..377c5f2d4 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -18,7 +18,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = STM32F103RE_btt
 
 [common]
 default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
~~~

greetings

sl1pkn07, 2020-04-15:
~~~
  // Allow language selection from menu at run-time (otherwise use LCD_LANGUAGE)
  //#define LCD_LANGUAGE_1 en
  //#define LCD_LANGUAGE_2 fr
  //#define LCD_LANGUAGE_3 de
  #define LCD_LANGUAGE_4 es
  //#define LCD_LANGUAGE_5 it
~~~

~~~
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/language_menu.cpp: In static member function 'static void LanguageMenu::onRedraw(draw_mode_t)':
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/language_menu.cpp:43:50: error: 'GET_LANGUAGE_NAME' was not declared in this scope
   cmd.tag(1).button(BTN_POS(1,1), BTN_SIZE(1,1), GET_LANGUAGE_NAME(1));
                                                  ^~~~~~~~~~~~~~~~~
Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/stepper_bump_sensitivity_screen.cpp.o
Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/language_menu.cpp:43:50: note: suggested alternative: 'LCD_LANGUAGE_3'
   cmd.tag(1).button(BTN_POS(1,1), BTN_SIZE(1,1), GET_LANGUAGE_NAME(1));
                                                  ^~~~~~~~~~~~~~~~~
                                                  LCD_LANGUAGE_3
~~~

same error when add some languages

marciot, 2020-04-16:
> with hardware SPI
>
> ```
> Compiling .pio/build/STM32F103RE_btt/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/base_numeric_adjustment_screen.cpp.o [...]

It looks like the function for transferring multiple bytes is missing in the STM32F1 HAL. Someone needs to write it.

marciot, 2020-04-16:
> Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/language_menu.cpp:43:50: error: 'GET_LANGUAGE_NAME' was not declared in this scope
> same error when add some languages

@sl1pkn07: I originally wrote the multi-language support but it has received very little testing. Looks like some thing got lost in translation (i.e. some code got lost). See commit 6126c9b for the fix.

GerogeFu, 2020-04-16:
> hardware conected and not works :/
> 
> anyone have the pinout of this screen? @GerogeFu ?

I just read my email , here is the info : https://github.com/FYSETC/TFT81050

sl1pkn07, 2020-04-16:
very thanks @GerogeFu for the schemas/specs!  but unfortunately, the config exposes in https://github.com/MarlinFirmware/Marlin/pull/17568 is the same config as `LCD_ALEPHOBJECTS_CLCD_UI` have, which i have test and i get this

[image]
[image]

@marciot i also test your octupus examples. all works except the UTF8 example 

< offtopic>

<details>

~~~
In file included from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../../basic/ftdi_basic.h:35:0,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../ftdi_extended.h:26,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/western_char_set.cpp:22:
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
   #if !HAS_RESOLUTION
        ^~~~~~~~~~~~~~
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
In file included from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../../basic/ftdi_basic.h:35:0,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../ftdi_extended.h:26,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/standard_char_set.cpp:22:
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
   #if !HAS_RESOLUTION
        ^~~~~~~~~~~~~~
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/../../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
In file included from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/ftdi_basic.h:35:0,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/ftdi_extended.h:26,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/event_loop.cpp:23:
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
   #if !HAS_RESOLUTION
        ^~~~~~~~~~~~~~
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
In file included from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/ftdi_basic.h:35:0,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/spi.cpp:23:
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
   #if !HAS_RESOLUTION
        ^~~~~~~~~~~~~~
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
In file included from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/ftdi_basic.h:35:0,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/commands.cpp:23:
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
   #if !HAS_RESOLUTION
        ^~~~~~~~~~~~~~
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
In file included from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/ftdi_basic.h:35:0,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/ftdi_extended.h:26,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/sound_player.cpp:22:
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
   #if !HAS_RESOLUTION
        ^~~~~~~~~~~~~~
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/boards.h:164:8: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
In file included from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/../basic/ftdi_basic.h:35:0,
                 from /tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/ftdi_extended.h:26,
[... cut ...]
/tmp/arduino_build_349971/sketch/src/ftdi_eve_lib/extended/unicode/western_char_set.cpp:434:34: warning: enumeral and non-enumeral type in conditional expression [-Wextra]
       base_char   = base_special ? NO_DOT_I : std_char;
                     ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
~~~

</details>
if comment `define TOUCH_UI_UTF8_WESTERN_CHARSET` (and the thig SPI_SPEED) builds ok

tested with the octopus and bugfix-2.0.x ftdi_eve library

but the same config in marlin works ok

< /offtopic>

greetings

RudolphRiedel, 2020-04-16:
> I just read my email , here is the info : https://github.com/FYSETC/TFT81050
@GerogeFu Please post the exact modell of the TFT in use or even better a datasheet for it.

sl1pkn07, 2020-04-16:
> This might be working better, this is a lot closer to the 20+ displays I support with my library:
> 
> ```c++
> #elif defined(TOUCH_UI_800x480)
>   namespace FTDI {
>     constexpr uint8_t Pclk                 =    2;
>     constexpr uint8_t Pclkpol              =    1;
>     constexpr uint16_t Hsize               =  800;
>     constexpr uint16_t Vsize               =  480;
>     constexpr uint16_t Vsync0              =    0;
>     constexpr uint16_t Vsync1              =    2;
>     constexpr uint16_t Voffset             =   32;
>     constexpr uint16_t Vcycle              =  525;
>     constexpr uint16_t Hsync0              =    0;
>     constexpr uint16_t Hsync1              =   48;
>     constexpr uint16_t Hoffset             =   88;
>     constexpr uint16_t Hcycle              =  928;
> 
>     constexpr uint32_t default_transform_a  =  0x0000D8B9;
>     constexpr uint32_t default_transform_b  =  0x00000124;
>     constexpr uint32_t default_transform_c  =  0xFFE23926;
>     constexpr uint32_t default_transform_d  =  0xFFFFFF51;
>     constexpr uint32_t default_transform_e  =  0xFFFF7E4F;
>     constexpr uint32_t default_transform_f  =  0x01F0AF70;
>   }
> ```

Hi, i have test this

~~~patch
diff --git a/Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/resolutions.h b/Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/resolutions.h
index 471530cad..a89009ff7 100644
--- a/Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/resolutions.h
+++ b/Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/resolutions.h
@@ -43,8 +43,8 @@
     constexpr uint16_t Vcycle               = tv; \
     constexpr uint16_t Vsync0               = tvfp - 1; \
     constexpr uint16_t Vsync1               = tvfp + tvpw - 1; \
-    static_assert(thfp + thb + Hsize == th, "Mismatch in display th"); \
-    static_assert(tvfp + tvb + Vsize == tv, "Mismatch in display tv");
+    /*static_assert(thfp + thb + Hsize == th, "Mismatch in display th"); \
+    static_assert(tvfp + tvb + Vsize == tv, "Mismatch in display tv"); */
 
 #ifdef TOUCH_UI_320x240
   namespace FTDI {
@@ -98,20 +98,20 @@
 
 #elif defined(TOUCH_UI_800x480)
   namespace FTDI {
-    constexpr uint8_t  Pclk                 =    3;
+    constexpr uint8_t  Pclk                 =    2;
     constexpr uint8_t  Pclkpol              =    1;
     constexpr uint16_t Hsize                =  800;
     constexpr uint16_t Vsize                =  480;
 
-    constexpr uint16_t th                   = 1056; // One horizontal line
-    constexpr uint16_t thfp                 =  210; // HS Front porch
-    constexpr uint16_t thb                  =   46; // HS Back porch (blanking)
-    constexpr uint16_t thpw                 =   23; // HS pulse width
+    constexpr uint16_t th                   =  928; // One horizontal line
+    constexpr uint16_t thfp                 =    1; // HS Front porch
+    constexpr uint16_t thb                  =   88; // HS Back porch (blanking)
+    constexpr uint16_t thpw                 =   48; // HS pulse width
 
     constexpr uint16_t tv                   =  525; // Vertical period time
-    constexpr uint16_t tvfp                 =   22; // VS Front porch
-    constexpr uint16_t tvb                  =   23; // VS Back porch (blanking)
-    constexpr uint16_t tvpw                 =   10; // VS pulse width
+    constexpr uint16_t tvfp                 =    1; // VS Front porch
+    constexpr uint16_t tvb                  =   32; // VS Back porch (blanking)
+    constexpr uint16_t tvpw                 =    2; // VS pulse width
 
     COMPUTE_REGS_FROM_DATASHEET

 ~~~
i can't see any visible difference. is a cosmetic change more than funcional change?

greetings

RudolphRiedel, 2020-04-16:
>i can't see any visible difference.

Hmm, okay.

>is a cosmetic change more than funcional change?

It is a functional change if the display timings are changed to what they are supposed to be.
This is why  I asked for the datasheet of the panel that FYSECT is using.
But then I also do not have this data for the Lulzbot display.

Here have a look at the 800x480 display timings for EVE modules that I gathered so far:

[Display_Parameter_800x480.xlsx](https://github.com/MarlinFirmware/Marlin/files/4489128/Display_Parameter_800x480.xlsx)

The only one really sticking out there is the one from Marlin.

sl1pkn07, 2020-04-16:
About lulzbot tft

https://ohai.lulzbot.com/project/touch-screen-assembly/quiver/
> 1x [EL-MS0552] Display, 5" BOE VTFT050MRNT01, WVGA TFT with resistive touch panel

[image]

but i can't find the datasheet :(. 

I've contact with BOEVX for if can send me the datasheet, but i think is not to be success :/

greetings

RudolphRiedel, 2020-04-16:
In any case, the display parameters look odd to me but they are not exactly wrong or otherwise it would look not as nice as it does.
That is not the real issue.

thinkyhead, 2020-04-17:
> It looks like the function for transferring multiple bytes is missing in the STM32F1 HAL. Someone needs to write it.

It looks like the SPI classes call this `write` and not `transfer`.

RudolphRiedel, 2020-04-17:
I even have more basic issues trying to compile with "#define TOUCH_UI_FTDI_EVE" in Configuration.h active:

In file included from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\archim2-flash\../ftdi_eve_lib/extended/ftdi_extended.h:47:0,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\archim2-flash\../ftdi_eve_lib/ftdi_eve_lib.h:27,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\archim2-flash\flash_storage.cpp:27:
Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\archim2-flash\../ftdi_eve_lib/extended/sound_list.h: In static member function 'static const char* SoundList::name(uint8_t)':
Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\archim2-flash\../ftdi_eve_lib/extended/sound_list.h:33:29: error: 'pgm_read_ptr_near' was not declared in this scope        
       return (const char* ) pgm_read_ptr_near(&list[val].name);

My target is STM32F103RC_fysetc and not AVR, so it should not even try to use any of the AVRlibc memory functions.

sl1pkn07, 2020-04-17:
in that error I set this in the HAL.h file

~~~patch
diff --git a/Marlin/src/HAL/STM32F1/HAL.h b/Marlin/src/HAL/STM32F1/HAL.h
index ff42beb92..48c9e8605 100644
--- a/Marlin/src/HAL/STM32F1/HAL.h
+++ b/Marlin/src/HAL/STM32F1/HAL.h
@@ -47,6 +47,9 @@
   #include "msc_sd.h"
 #endif
 
+
+#define pgm_read_ptr_near(addr) pgm_read_ptr(addr)
+
 // ------------------------
 // Defines
 // ------------------------
~~~

(if see the parameters in $HOME/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/avr/pgmspace.h, all of them uses (addr)

RudolphRiedel, 2020-04-17:
I changed that pgm_read_ptr_near() in sound_list.h to pgm_read_ptr_far().
That pgm_read_ptr_near() would be wrong even for AVR, or which 64k AVR runs MARLIN?

sl1pkn07, 2020-04-17:
i'm not coder, so idk

greetings

RudolphRiedel, 2020-04-17:
pgm_read_ptr_near() is for 16 bit adresses

RudolphRiedel, 2020-04-17:
Hmm, next issue, the EXP1 header on my board is connected to SPI_2 but it looks like TOUCH_UI_FTDI_EVE is hardcoded to use SPI_1.

RudolphRiedel, 2020-04-17:
Okay, I modified spi.cpp and spi.h from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\basic a little: [Marlin_EVE_SPI.zip](https://github.com/MarlinFirmware/Marlin/files/4493278/Marlin_EVE_SPI.zip)

I put an instance of the SPIClass into the FTDI namespace that is initialised to the spiPortNumber defined with CLCD_SPI_BUS.
And then I changed a bunch of calls to use EVE_SPI instead of ::SPI.

Oh yes, I also changed it to use .write().

I am just hacking away here, I am in no way confident enough to put this in a PR.
And instancing the class like this will probably only work with the STM32 HAL, with the special version of the STM32 HAL that is included in MARLIN.
The Arduino core from ST is using a different constructor, so is the SAMD core.

I also added this to my pins_FYSETC_CHEETAH.h:
#if ENABLED(TOUCH_UI_FTDI_EVE)
    #define CLCD_SPI_BUS    2
    #define CLCD_MOD_RESET  PC9
    #define CLCD_SPI_CS     PB12

    //#define CLCD_USE_SOFT_SPI
    #if ENABLED(CLCD_USE_SOFT_SPI)
      #define CLCD_SOFT_SPI_MOSI PB15
      #define CLCD_SOFT_SPI_MISO PB14
      #define CLCD_SOFT_SPI_SCLK PB13
    #endif
#endif

And at least my Logik Analyzer is showing 8MHz traffic now.

I still need to hook up a display to my board though to see if it works.

RudolphRiedel, 2020-04-17:
I tried both an EVE2_50G and an EVE3_50G and they are both displaying the main screen.
Both do flicker a little though.
And I do not have touch yet as both of these use GT911 capacitive touch controllers.

But I have a Marlin Bugfix 2.0x working with a 800x4800 TOUCH_UI_FTDI_EVE display running on a FYSETC Cheetah V1.1 which is a lot closer to the TFT81050 and the FYSETC S6 than using some test-code on an Arduino.

RudolphRiedel, 2020-04-17:
Okay, I replaced the timing settings and the flickering is gone.

I activated the GT911 touch controller for the EVE3-50G and now I can use the GUI.

Is TOUCH_UI_FTDI_EVE really supposed to be in working condition?
Well, maybe the issues I have are because I only have the display connected to the board and nothing else.

But a couple of things do look a bit strange.
Apart from the wrong font-size for the "Sensorless Homing" button on the "Advanced Settings" menu, using the button "Interface" makes it hang.

[image]
[image]
[image]
[image]

I just switched to Software-SPI and that is not working at all, hmm.

marciot, 2020-04-17:
@RudolphRiedel: Yeah, that looks pretty messed up. The portrait mode with a 800x480 resolution is the only one I regularly test and the only one that is used commercially. This means the other orientation and resolutions often get messed up, which seems to be what happened here.

sl1pkn07, 2020-04-17:
then only works with hardware SPI? because with you changes, i'm stuck in my first image i posted (deface background, only show 2 buttons)

the SKR E3 DIP uses the hardware SPI for stepers and the SD

i need set this  `#define CLCD_SPI_BUS 2` ? what value i need set in my case?

greeetinsg and wreat work!

RudolphRiedel, 2020-04-17:
It just got a whole lot more interesting:
[image]

@marciot 
Okay, I check the portrait mode.
When switching back to hardware SPI I found out that my FFC cable was defective.
And after I replaced it with a fresh one this is what I get with software SPI.

marciot, 2020-04-17:
@RudolphRiedel: Those kinds of artifacts often happen when the FTDI does not have enough time to go through the display list during each scanline. Reducing the display clock rate will solve that problem.

marciot, 2020-04-17:
I don't have time right now, but I'll try fixing the landscape mode later. What resolution screen are you using?

RudolphRiedel, 2020-04-17:
> @RudolphRiedel: Those kinds of artifacts often happen when the FTDI does not have enough time to go through the display list during each scanline. Reducing the display clock rate will solve that problem.

I never encountered these types of artifacts before.
And apart from that there is not really much happening there, it works with hardware SPI using the exact same display settings.

The screen has 800x480 and so far I am a bit struggeling to find the switch for portrait mode.

[119 more replies]

commit 791873b1 (2020-04-16): Fix ExtUI compile errors (#17544)

Fixes #16628
closed 2020-05-13 by thinkyhead