MarlinFirmware/Marlin #16629: AUTO_BED_LEVELING_BILINEAR  uses wrong area limits

bill-orange, 2020-01-20:
### Bug Description
Regarding Tronxy X3A with configuration only lightly modified from the sample files:

AUTO_BED_LEVELING_BILINEAR correctly probes a grid and probes it the correct number of times.  The Grid, however, is only a small square roughly a quarter of the bed size.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4087972/Marlin.zip)

### Steps to Reproduce

1.  Enable AUTO_BED_LEVELING_BILINEAR and probe the bed
2. observe the small pattern.

**Expected behavior:

Whole bed probed

**Actual behavior:*

Only small area probed.

#### Additional Information

The 2.0.x bugfix version had around line 1055 in Configuration.h a section that allowed the probe limits to be defined.  I do not see a similar section on 2.0.1  Without this section I can not fix the problem.

From bugfix:

```
#define LEFT_PROBE_BED_POSITION MIN_PROBE_EDGE
#define RIGHT_PROBE_BED_POSITION (X_BED_SIZE - MIN_PROBE_EDGE)
#define FRONT_PROBE_BED_POSITION MIN_PROBE_EDGE
#define BACK_PROBE_BED_POSITION (Y_BED_SIZE - MIN_PROBE_EDGE)

```

sjasonsmith, 2020-01-20:
What code are you running? This sounds like an issue that was already fixed. I believe it made it in to 2.0.1.

bill-orange, 2020-01-20:
Oops. Left that out of my post. I am on 2.0.1

Vertabreak, 2020-01-20:
[image]
the defines your looking for should be in configuration_adv.h they may have been moved.

bill-orange, 2020-01-20:
I am not sure why they would be commented out as the default since they are in the if statement,  In any case, commenting them makes no difference.  The default value appears to be 30 which is a reasonable number.

sjasonsmith, 2020-01-21:
> Oops. Left that out of my post. I am on 2.0.1

I think I was wrong thinking that was in 2.0.1. I lose track of the timelines for the labels since I'm in the bugfix branch all the time. You should try again with the latest bugfix-2.0.x branch.

You are most likely experience the issue fixed here:
#16367 

> The default value appears to be 30 which is a reasonable number if it worked.

30 is really high for most printers. Your NOZZLE_TO_PROBE_OFFSET is already factored in, MIN_PROBE_EDGE is just extra distance to stay away from the edge of the bed. It is useful if you have a probe which does't work close to the bed edge, or have something like clips that you need to avoid.

bill-orange, 2020-01-21:
Since this issue has been ( or is being) addressed at least two other places, I will close this issue.  I will most likely wait for a release with this issue fixed.  I can limp along for now.

closed 2020-01-21 by bill-orange