MarlinFirmware/Marlin #16630: Ender 5 Z axis Steps set to high by default

jasonelmore, 2020-01-20:
# NO SUPPORT REQUESTS PLEASE

Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

jasonelmore, 2020-01-20:
Steps for the Z axis are set to 800 steps per mm using the configuration files for Marlin 2.0.1, when it should be set to 400.  Settings for flow rate, acceleration and jerk are also off.

boelle, 2020-01-20:
@thinkyhead you have time to look at this one?

Vertabreak, 2020-01-20:
iirc they changed the z rod type on that printer so some are 800 and some are 400.
but all of the above noted settings can easily be configured on the LCD of by gcode in terminal.
with all the variations in what is listed the same model from the same company is becomes very hard to cover all the bases. at some point where do you draw the line between what can be fine tuned by the user and what an example should be. one example is i have an A10 and i can verify there are no less them 100 variations of this printer in the wild right now they make a change every bath and ship it out and make no listing they did anything anywhere. not to try to discount anyone's issue but i want it to be known how nearly impossible a task the company's make it to offer perfect examples for each model.

thinkyhead, 2020-01-21:
The solution is to make the steps-per-unit undefined in the example config and add a sanity-check (right in the config file) that warns the user to choose the one with 400 or the one with 800.

thisiskeithb, 2020-01-21:
> Steps for the Z axis are set to 800 steps per mm using the "Ender 5) configuration files included with Marlin 2.0.1, when it should be set to 400. Settings for flow rate, acceleration and jerk are also off.

[It's set to 400 in the default Ender-5 config](https://github.com/MarlinFirmware/Configurations/blob/decc096985447bb64064f7e59a64a9a1f3bcb943/config/examples/Creality/Ender-5/Configuration.h#L728): `#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 93 }` and there's a [readme](https://github.com/MarlinFirmware/Configurations/blob/decc096985447bb64064f7e59a64a9a1f3bcb943/config/examples/Creality/Ender-5/README.md) explaining the leadscrew change & how to set your steps to 800 if you have the new model.

> The solution is to make the steps-per-unit undefined in the example config and add a sanity-check #error (right in the config file) that warns the user to choose the one with 400 or the one with 800 before continuing.

This is probably the best solution moving forward since it'll make the user chose one.

jasonelmore, 2020-01-21:
Thanks for the clarification. Closing the issue now.

closed 2020-01-21 by jasonelmore