MarlinFirmware/Marlin #16631: Auto bed level MIN_PROBE_EDGE issue

onkytonk, 2020-01-21:
I am running an SKR e3 Mini v1.2 on my Ender 3 pro. 
I have compiled Marlin 2.0.1 and most functions are working correctly. 

I am trying to set the Min Probe edge to have a more symmetrical probing grid. 
However, it seems that the below function is DOUBLING the value for some reason. 

My probe offset is X=-45/Y=-10. 

When I have MIN_PROBE_EDGE_LEFT set to 15, the first probing point occurs at X=75 (15x2+45).
This multiplication occurs for any of the MIN_PROBE_EDGE values. 

I can't seem to get to the bottom of this. 

```
#if PROBE_SELECTED && !IS_KINEMATIC
  #define MIN_PROBE_EDGE_LEFT 15
  #define MIN_PROBE_EDGE_RIGHT 0
  #define MIN_PROBE_EDGE_FRONT 10
  #define MIN_PROBE_EDGE_BACK 10
```

thisiskeithb, 2020-01-21:
Give `bugfix-2.0.x` a try. That should have been fixed in #16367.

sjasonsmith, 2020-01-21:
> Give `bugfix-2.0.x` a try. That should have been fixed in #16367.

Thanks from saving me from looking this up. I thought that fix was already in 2.0.1, but I guess it came later. There's another open issue that looks like it as well.

bill-orange, 2020-01-21:
Same issue as mine #16629

boelle, 2020-01-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2020-01-21:
I am assuming this issue is already fixed, unless someone confirms they have problems using the latest state of bugfix-2.0.x.

boelle, 2020-01-23:
closing as assumed fix

we can reopen later if OP decides to come alive and post configs

closed 2020-01-23 by boelle

onkytonk, 2020-01-29:
Sorry, didn’t get any notifications that there were posts. 
I did attach a zip when I posted that included my configuration.h and configuration_adv.h. Don’t know what happened to it. 

Seems like the issue was resolved in the 2.0.x bugfix.
2.0.1 does not have the fix. I made the changes manually and it works s expected now.