MarlinFirmware/Marlin #16632: AUTO_BED_LEVELING_BILINEAR Not Reaching X or Y End of Bed

kgrigio, 2020-01-21:
### Bug Description

I am attempting to migrate from Marlin 1.1.5 to Marlin 2.0.x.  I can't seem to get the auto bed leveling working properly and I believe the issue is related to the fact that there isn't a LEFT_PROBE_BED_POSITION, RIGHT_PROBE... or a X_PROBE_OFFSET_FROM_EXTRUDER for me to define and this is what is throwing me off.

In Marlin 1.1.5 the auto bed level would go all the way to the x and y max positions (x 176, y 248), but now I am only going to x of 134 and a y of 223.

In Marlin 2.0.x I have set NOZZLE_TO_PROBE_OFFSET { 32, 5, -1.5 }, which would correlate to my X_PROBE_OFFSET_FROM_EXTRUDER, Y_PROBE... in Marlin 1.1.5, but this isn't fixing the bed leveling issue.

### My Configurations

I am attaching the configuration.h and configuration_adv.h first from Marlin 2.0.x
[Marlin2_0_x.zip](https://github.com/MarlinFirmware/Marlin/files/4088928/Marlin2_0_x.zip)

Next from my Marlin 1.1.5
[Marlin1_1_5.zip](https://github.com/MarlinFirmware/Marlin/files/4088932/Marlin1_1_5.zip)

Any help would be greatly appreciated.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-01-21:
```
// Certain types of probes need to stay away from edges
#define MIN_PROBE_EDGE 10

```
going to x of 134 + 32 (probe offset)  +10 = 176 (limit your seeing)
going to y of 223 + 5 (probe offset) +10 = 238   (doesn't work out..)   
A typo perhaps goes to 233 + 5 + 10 = 248  (limit your seeing)

boelle, 2020-01-21:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-01-21 by boelle

tpruvot, 2020-01-21:
in general MIN_PROBE_EDGE is to avoid the bed PEI clips, but UBL need specific MESH_INSET in config, to act like ABL...