MarlinFirmware/Marlin #16633: [BUG] UBL terminate with debug msg Unknown command: ""

EugeneSkrebnev, 2020-01-21:
### Bug Description
i am running UBL with 6 by 6 grid in the middle UBL terminates on 17 point.
i define DEBUG_LEVELING_FEATURE and i see 
echo:Unknown command: ""

### My Configurations
[Configuration_.zip](https://github.com/MarlinFirmware/Marlin/files/4090269/Configuration_.zip)
### Steps to Reproduce

run
>>> G28
SENDING:G28
>>> G29 P1
SENDING:G29 P1

UBL G29 P1 doesnt matter with cold or hot bed

**Expected behavior:** [What you expect to happen]
UBL success finish
**Actual behavior:** [What actually happens]
terminate with strange msg

#### Additional Information
- small beep when error occurs

log:
Connecting...
Printer is now online.
>>> G28
SENDING:G28
>>> G29 P1
SENDING:G29 P1
Probing mesh point 2/36.
Probing mesh point 3/36.
eho:busy:pocsin
ehobuy pcsig
Pobingmehpon /3.
Probing mesh point 5/36.
Probing mesh point 6/36.
Probing mesh point 7/36.
Probing mesh point 8/36.
Probing mesh point 9/36.
Probing mesh point 10/36.
Probing mesh point 11/36.
Probing mesh point 12/36.
Probing mesh point 13/36.
Probing mesh point 14/36.
Probing mesh point 15/36.
Probing mesh point 16/36.
Probing mesh point 17/36.
echo:Unknown command: ""

Vertabreak, 2020-01-21:
#define MIN_PROBE_EDGE 20 
will result in skipped rows on all sides maybe even more then one row seems to be working as expected.
#define MESH_INSET 7 
inset is also in use in the past noted odd behavior when both inset and min probe edge where used together
would suggest picking a single system to use for avoiding the edge of the bed, inset tends to be the most ideal method with UBL as it probes all points just in a smaller space rather then skipping entire rows like min probe edge.

EugeneSkrebnev, 2020-01-21:
maybe it would be good to have compile time defines hints to avoid errors in MIN_PROBE_EDGE vs MESH_INSET configuration

Vertabreak, 2020-01-21:
it was suggested before but i guess it wasn't how they wanted to handle it.

EugeneSkrebnev, 2020-01-21:
big thanks, that works, maybe that will somehow fix 16624 also

closed 2020-01-21 by EugeneSkrebnev

thinkyhead, 2020-01-21:
> it was suggested before but i guess it wasn't how they wanted to handle it.

Let's discuss what is actually going on before posting things that might start rumors.