MarlinFirmware/Marlin #16634: [BUG] Auto Level sets z home ~0.8mm below true home

jdelpero, 2020-01-21:
### Bug Description
After an auto home (g28) z home is correct and nozzle is ~0.1mm above bed when z is at 0. After an auto level routine (g29) z home is no longer correct and is at approx. -0.8mm below the original home point. Tested with both 3 Point & Bilinear bed levelling options, both with the same result.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4090945/Marlin.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Pre-heat for PLA
2. Run Auto Home, move nozzle to probe point and check z=0 is correct (paper test etc.)
3. Run Auto Level, move nozzle to auto home probe point and check z which is no longer correct and is now at +0.8mm (z home has moved below the bed)

**Expected behavior:** [What you expect to happen]
G29 auto levels bed and honours z probe offset

**Actual behavior:** [What actually happens]
G29 increases z probe offset by -0.8mm

#### Additional Information
Bed Mesh Results (g29)
      0      1      2
 0 -1.058 -0.968 -0.797
 1 -1.085 -1.032 -0.871
 2 -1.111 -1.108 -0.961

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

reloxx13, 2020-01-22:
duplicate
https://github.com/MarlinFirmware/Marlin/issues/15450

boelle, 2020-01-23:
closing since this is a dublicate of #15450

@jdelpero we only need one issue open for each issue

closed 2020-01-23 by boelle