MarlinFirmware/Marlin #16635: Bl touch stow/deploy pin translation error _es

davidcgu, 2020-01-21:
### Bug Description

Translation for stow/deploy pin for Bltouch is swapped on Spanish lang file: \Marlin\src\lcd\language\language_es.h

Lines 348&349   swapped "Subir/Bajar"

### Steps to Reproduce

When operating Bltouch on test mode from LCD the real behaviour is opposite than expencted.
That drives to confusion on Spanish lang users.

**Expected behavior:** [What you expect to happen] Already explained

**Actual behavior:** [What actually happens] Already Explained

Regards

boelle, 2020-01-21:
it happens faster if you submit a PR to bugfix 2.0.x branch

davidcgu, 2020-01-21:
Yep, I see, sure it was better choice!

As is notified already on PR you can delete or close this one if you wanted.

Thanks for your time

PR #16637 (2020-01-26): Fix Spanish bltouch reversed messages
Fix #16635
closed 2020-01-22 by boelle