MarlinFirmware/Marlin #16638: TMC connection error SSD1306 TMC2209 Re-arm

RasTschubai, 2020-01-21:
I have some trouble with my small I2S OLED SSD1306. When enabled in Marlin 2.0.1 it shows up great *buuuut* with a bottom line saying TMC CONNECTION ERROR. `M122`  in pronterface shows:
```
Driver registers:
		X	0xC0:0C:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0x00:00:00:00	 Bad response!
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... Error: All LOW
```
Disabling ssd1306 in **configuration.h**:
```
//#define U8GLIB_SSD1306
```
and everything works fine!
I'm using TMC2209 for all 4 stepper on a RAMPS 1.6PLUS with a Re-Arm clone. Tried different ports for E0 UART without success.

boelle, 2020-01-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

RasTschubai, 2020-01-22:
Sorry, you're right!
I also added the `pins_RAMPS_RE_ARM.h`. I think I didn't changed any files else.
Thank you!
[Configuration.h.gz](https://github.com/MarlinFirmware/Marlin/files/4096708/Configuration.h.gz)
[Configuration_adv.h.gz](https://github.com/MarlinFirmware/Marlin/files/4096709/Configuration_adv.h.gz)
[pins_RAMPS_RE_ARM.h.gz](https://github.com/MarlinFirmware/Marlin/files/4096711/pins_RAMPS_RE_ARM.h.gz)

boelle, 2020-01-28:
have you tried with bugfix 2.0.x ? that one is newer than 2.0.1

and its updated almost dayly so will 99.99% of the time be the best to test with

RasTschubai, 2020-01-29:
Thanks for your help, boelle!
I just tried latest bugfix 2.0.x and Marlin 2.0.2.
Unfortunately, without any changes in the behavior.

boelle, 2020-01-30:
2.0.2 and bugfix are broken right now so hold off for maybe a week or 2

RasTschubai, 2020-02-02:
...and Marlin 2.0.3 doesn't work either.
I had one TMC2130 lying around. I installed it in standalone mode. This works for now with the SSD1306 and is sufficient.
Right now I'm struggling to get a separate  rotary encoder to work... But that's a different story.

flankerzo, 2020-02-06:
can you please explain how to connect / hardware side/ ssd1306 ? i have both, SPI and UART, i have MKS SGEN L using 5x TMC2130 via SPI, no senseless homing, Thanks

RasTschubai, 2020-02-07:
The SSD1306 is a I2C display. I couldn't see any I2C connectors on the MKS SGEN L board. But I don't know if there are any hacks around. The L1768/1769 obviously has I2C capabilities.

boelle, 2020-02-08:
with the re-arm you use a ramps board on top of it, that is what it is made for, to be a drop in replacement for the mega2560 board

[image]

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle