MarlinFirmware/Marlin #16643: [BUG] Multiple HW platforms with BLTouch read triggered

ebraiman, 2020-01-21:
### Bug Description

After configuring Marlin bugfix 2.0.x for BLTouch M119 reads Triggered before homing z-axis.  

[Marlin-bugfix-2.0.x_SKR1_4.zip](https://github.com/MarlinFirmware/Marlin/files/4094384/Marlin-bugfix-2.0.x_SKR1_4.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Hardware:
Acer Nitro 5 with Windows 10 64 bit
SKR 1.4 (LPC1768)
BLTouch v2.0/v3.1
USB cable

Software:
VScode with platofrmIO IDE plugin
Marlin Bugfix 2.0.x

Hardware configuration:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/BTT%20SKR%20V1.4%20Instruction%20Manual.pdf 

Marlin Changes
Configuration.h:

change to, “#define SERIAL_PORT -1”
change to, “#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_4”
remove comment for, “define BLTOUCH”
remove comment for, “#define NUM_SERVOS 3”.
change to, “#define NUM_SERVOS 1”.

PlatformIO.ini changes:
default_envs = LPC1768

Click upload in VSCode
After upload checked drive for firmware.bin

PO/PO USB cable

Open Pronterface and connected to SKR 1.4 (LPC1768)
Entered, “M119”, in Pronterface.
**Expected behavior:** [What you expect to happen]
SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: open

**Actual behavior:** [What actually happens]
SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
This occurs on the following platforms:
MEGA2560/RAMPS 1.4
MEGA2560/RAMPS 1.6
SKR 1.3 (LPC1768)
SKR 1.4 (LPC1768)
SKR 1.4 (LPC1769)
SKR PRO v1.1
using the following touch sensors:
BLTouch v3.1
BLTouch v2.0
3D Touch
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ebraiman, 2020-01-21:
Additional Links:
https://github.com/MarlinFirmware/Marlin/issues/16469
https://github.com/MarlinFirmware/Marlin/issues/16552

ebraiman, 2020-01-21:
Does anyone know the correct new steps to configure or possible other connected HW needed such as stepper boards, Hotends, thermistors, etc?

ebraiman, 2020-01-22:
Updating instructions, as the appear to be missing commented out, "#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN"

ebraiman, 2020-01-22:
Connecting...
Printer is now online.
>>> M119
SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED
z_probe: TRIGGERED

ebraiman, 2020-01-22:
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4094545/Marlin-bugfix-2.0.x.zip)
Here is the update.

ebraiman, 2020-01-22:
This also occurs on SKR 1.4 with z-min pins.

GMagician, 2020-01-22:
@ebraiman is it `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` enabled or not? sorry not clear to me due to required translation to understand (English is not my language) and also because your attached config has been posted after but with option commented out.

AmiSMB, 2020-01-22:
On my printers that are working with BLTouch I have Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled. If I turn one the printers and connect Z will show as triggered when issuing an M119.
I have found that if I deploy the BLTouch:-
  M280 P0 S10
then issue a
  M119
I see
  z_min: open

If I stow the probe issuing
  M280 P0 S90
then issue a 
  M119
I see
  z_min: TRIGGERED

which is the way that it is working in Marlin 2.0.x

tatusah, 2020-01-22:
You mean @ebraiman that you are worried because the probe shows triggered status when the probe pin is stowed? But the homing/auto-leveling itself works for you?

I personally much prefer that it's triggered while stowed as it should actually make detecting the probe touch more reliable because instead of short signal the state change is "latching". This shouldn't affect functionality except probably if one would enable ENDSTOPS_ALWAYS_ON_DEFAULT and SD_ABORT_ON_ENDSTOP_HIT.

My two BLTouch probes, both v3.1 have always behaved like this. Always triggered in stowed position and open when deployed. This is at least while using the default BLTouch open drain mode which needs pull-up resistors enabled. This behavior of the probe state being reported always triggered in stowed position is result of the pull-up resistor. It also works as a safety feature if the probe would get disconnected then the state is triggered and the carriage doesn't hit the bed if you would try to home or do auto-leveling.

Haven't tried with the 5V mode as the endstop signal lines on my boards have 3.3V logic but possibly in 5V mode without pull-up sensors would show the triggered status only momentarily while the probe actually touches the bed. But again I think current "latching" behavior is better at least in my use case. So if your probe actually works for auto-leveling/homing then I would say everything is OK. If really want to see the trigger state as open while probe pin is stowed then if you have 5V logic board you can go around playing with pull-up/pull-down resistor settings.

ebraiman, 2020-01-22:
Originally it appeared open in the several tutorials ive created.  Now user needs to do the following:
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED
>>> M280 P0 S10
SENDING:M280 P0 S10
>>> M119
SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: open
echo:busy: processing
echo:busy: processing
Error:!! STOP called because of BLTouch error - restart with M999
[ERROR] Error:!! STOP called because of BLTouch error - restart with M999

>>> M119
SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED
keep in mind a Thermistor may need to be present to do this, but I'll check more after work.

ebraiman, 2020-01-22:
Understand a thermistor was not always present in the past tutorials.  And if you ask why one might not use a thermistor think about what may be other functionality the marlin firmware may be used for?  For instance a laser cutter, router cutter, water cutter, etc.

GMagician, 2020-01-22:
Going back in time I always seen 'open' with bltouch, when stowed and when deployed...
Now for sure it's open when deployed but I can't test when stowed because as soon as I execute M119 pin will deploy

AmiSMB, 2020-01-22:
That is strange as on my two printers with Marin 2.0.x one ReARM and one SKR E3 Mini DIP the BLTouch stays deployed when I do a M119 after asking it to be deployed with M280 P0 S10.

Here are my configurations that are working:-

[Ender3Pro_SKRE3MiniDIP_Marlin2.0.x_config.zip](https://github.com/MarlinFirmware/Marlin/files/4099044/Ender3Pro_SKRE3MiniDIP_Marlin2.0.x_config.zip)
[P3Steel_ReARM_RAMPS14_Marlin2.0.x_config.zip](https://github.com/MarlinFirmware/Marlin/files/4099045/P3Steel_ReARM_RAMPS14_Marlin2.0.x_config.zip)

tatusah, 2020-01-22:
@ebraiman regardless of what the state is when the probe pin is stowed does the probe work for homing/auto-leveling? I think the functionality with homing and auto-leveling is the main point. If the probes don't work with homing and auto-leveling, then there could also be the possibility that the two probes of yours have been damaged and are now misbehaving with the boards you are connecting them to.

@GMagician for me the probes have always (about 3 months in use now) reported the M119 status as triggered while stowed and open while deployed. I have seen others tell the same as did @AmiSMB just above. And because this way of functionality seems to work reliably with homing and auto-leveling I'm happy with it although I don't mind even if it was open after it was triggered as long as everything works. Newer had the probes fail during probing so far with this "latching" behavior. But certainly the fact that your probe is deploying with the command M119 is strange. Have you by any change had the BLTouch in 5V mode and the pull-up/-down resistor settings together with signal inverting in some other state previously when it reported open while stoved? In any case M119 doesn't make the pin stove or deploy on my systems with skr e3 mini v1.2 boards, probe connected to z-min and the servo port with BLtouch v.3.1 default open drain mode. I'll have to see how the SKR 1.4 behaves when it arrives.

What's the course of action here? @ebraiman check if the probes work with homing/leveling and we ask around if the "latching" behavior was introduced by some code change along the way and whether it can be made as an option if some prefer it and others don't?

GMagician, 2020-01-22:
@tatusah mine is a 2.0 clone, installed on a ramp with mega2560 so 5V is the only choice I have. Mine is working not perfectly but I can do G28 and G34 without any issues

tatusah, 2020-01-22:
Ok @GMagician does it have option to change between fixed 5V and the open drain mode? That open drain mode is convenient as it works both in 3.3 and 5 V systems and as such based on BLTouch documentation there shouldn't even be the need to change the mode to fixed 5V mode in most cases. Anyway perhaps the probe pin deploying with M119 on the clone could possibly be a reliability issue and not a bug in Marlin.

GMagician, 2020-01-22:
I think it has no 3.3 option nor drain mode (that is present on 3.0 AFAIK).
Problem I have is due to 3.0 integration into Marlin. Some commands have been added for that but my clone misunderstand them...This is an old story...but despite of this mine is working

ebraiman, 2020-01-23:
Previous behavior observed in my case was probe was in non-deployed mode when M119 was used to check endstop state.  I've recorded several tutorials with this state of open for Z-min.  Now it appears we need to add a deploy command to have a opened state on z-min.  One would think the previous observed behavior deployed on homing of z axis.

Here is an example with SKR 1.3 using z-min endstop for signal pin.

https://www.youtube.com/watch?v=5cSzFCv7K4Q&t=5s

Around 11:09 into video you'll see the test.

tatusah, 2020-01-23:
Setting those isolated tabletop test aside do you have some problem with the probe while homing or auto-leveling in actual use?

ebraiman, 2020-01-23:
Yes,  a M999, which seems related to thermistor.  I'll be placing in a setup to check if it works soon.

boelle, 2020-01-26:
@ebraiman any news?

anyone else been able to reproduce this one and can confirm it?

ebraiman, 2020-01-26:
Just order a new MEGA2560/RAMPS and BLtouch to test.  I'll make a video from box opening to test on video to verify.

tatusah, 2020-01-26:
If we are talking about the probe being reported triggered by M119 while the probe pin is in stowed position I think anyone with BLTouch and up to date bugfix branch Marlin can report the same behavior. Even if this seems to be changed at some moment based on @ebraiman videos this doesn't seem to cause any problems. If anything it could possibly make the probing action more reliable as the system reads latching signal instead of a short pulse.

There are the videos showing the board alone connected to the probe showing the M119 reporting the status for BLTouch as triggered while stowed. And M119 documentation https://marlinfw.org/docs/gcode/M119.html mentions that the BLTouch signal is brief and that triggered state indicates error but this documentation might be outdated and might also not take into consideration the possible pull-up resistors affecting the behavior.

My complete printers also show this behavior but it doesn't seem to affect the functionality of using the BLTouch for homing or auto-leveling. Actually it has been 100% reliable with sensing the probe pin touching the build plate. And at least personally I haven't seen any reports of this causing problems. I think the question still remaining is what would @ebraiman suggest to be done for this and if the M999 problem he mentioned was actually unrelated to this.

ebraiman, 2020-01-26:
I have a recorded uploading to YouTube in two parts because the camera may have powered down or froze.  It's a bit scatter brained of a videos, but get the point across as to the odd functionality across HW/SW platforms.  
Part 1
https://youtu.be/dS-n1iIr8XI
Part 2
https://youtu.be/XyoYOpt0qa0

ebraiman, 2020-01-27:
Just tested a 5v Inductive sensor https://www.amazon.com/gp/product/B073XDCR2F/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1
on SKR v1.4 and this has the same issue as BLTouch.  I'll retest sensor on RAMPS 1.4, but It worked earlier today.  Also test a 3 wire switch endstop which worked fine on SKR v1.4

tatusah, 2020-01-27:
Unless there is a fully build printer with some issues related to this matter I don't see value in continuing. There isn't currently any reported issues indicating that the current way of M119 reporting probe status would cause some actual problems on fully built machines. The provided videos show USB powered bare boards on the table having just BLTouch or some other probe connected. This kind of testing is prone to display other unrelated issues as the USB might not power the board properly and the board is lacking the feedback from temp sensors and endstops unless these things are taken into consideration.

boelle, 2020-01-27:
i agree with @tatusah testing has to be done on a fully build/working printer that is assembled correctly

mtfisher, 2020-01-27:
Just to mention i reported a similar issue with a built printer

tatusah, 2020-01-28:
So with the built printer is the homing failing using the probe?

boelle, 2020-01-28:
@ebraiman can you provide pictures of how its connected?

GMagician, 2020-01-28:
> Just tested a 5v Inductive sensor https://www.amazon.com/gp/product/B073XDCR2F/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1

PNP sensor? sure it shouldn't be NPN? if your board has a pullup resistor a PNP sensor will never works

mtfisher, 2020-01-28:
Was able to get this working by fliping white and black cables on my jst 2 pin connector

boelle, 2020-01-28:
hehe, hence we need to see pictures from @ebraiman prob did the same mistake and it does not matter how its connected to endstop

ebraiman, 2020-01-28:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/BTT%20SKR%20V1.4%20Instruction%20Manual.pdf

The first wiring I tried was per the Manual.  The second was connected via ground for black and white for signal on z-Min using SKR 1.4.  I'll try reversing cable for both to see if any change occurs.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle