MarlinFirmware/Marlin #16644: [BUG + FIX] M43 reporting broken on stable 2.0.x on AVR uC

bsccara, 2020-01-22:
Using Pronterface to send a M43 command to my machine running stock 2.0.x the reply is always interrupted partway by a 

> 'Got rubbish reply from /dev/ttyUSB0 at baudrate 250000: Maybe a bad baudrate?'

error on Pronterface's log. Whatever the baudrate selected the error always shows up.
I've modified Pronterface's code to log the actual bytes received as reply and got this:

`b'PIN:   4   Port: G5        FIL_RUNOUT_PIN                         Output = 1    TIMER0B   PWM:   128    WGM: 3    COM0\x83: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow interrupt enabled  \n'`

So the problem is that on all COM fields Marlin is sending a trailing character above 128 and that causes a failure in decoding to ASCII on Pronterface.
I've tracked the problem down to the following function on HAL/HAL_AVR/pinsDebug.h:

`void com_print(uint8_t N, uint8_t Z) {`
`const uint8_t *TCCRA = (uint8_t*)TCCR_A(N);`
`SERIAL_ECHOPGM("    COM");`
`SERIAL_CHAR('0' + N);`  
`SERIAL_CHAR('A' + Z);`  
`SERIAL_ECHOPAIR(": ", int((*TCCRA >> (6 - Z * 2)) & 0x03));`  
`}`

Specifically the line 'SERIAL_CHAR('A' + Z);' must be '  SERIAL_CHAR(Z);' as the Z parameter is not an index but an actual character code.

ellensp, 2020-01-22:
or change the call com_print(T,L);  to com_print(T,L - 'A');  It is only used the once, so either way... 
But yes its a bug...

boelle, 2020-01-28:
@ellensp did PR 16645 fix this one?

ellensp, 2020-01-29:
yes its fixed

closed 2020-01-29 by boelle