MarlinFirmware/Marlin #16647: [BUG] (short description)

maukcc, 2020-01-22:
### Bug Description

If PINS_DEBUGGING is defined, and any serial port other then 0 is chosen, we get the following error when compiling:

> sketch/src/gcode/config/../../pins/pinsDebug.h:142:9: error: 'RXD_NAME_0' was not declared in this scope
>        { RXD_NAME_0, 0, true },
>          ^
> sketch/src/gcode/config/../../pins/pinsDebug.h:143:9: error: 'TXD_NAME_0' was not declared in this scope
>        { TXD_NAME_0, 1, true },

### My Configurations

bugfix-2.0.x
#define MOTHERBOARD BOARD_CNCONTROLS_15D

ellensp, 2020-01-22:
Need full configuration files.   Can't reproduce.

ellensp, 2020-01-22:
Managed to reproduce, that is odd... only happens on due based boards...   just testing a fix.

ellensp, 2020-01-22:
Issue is SERIAL_PORT_2 code is being compiled in when its not defined in Configuration.h, SERIAL_PORT_2 defaults to serial0 so expects  'RXD_NAME_0 and 'TXD_NAME_0 to be defined. Breaks when SERIAL_PORT is anything but 0.  The fix is ready, just waiting for it to be merged.

ellensp, 2020-01-26:
This has been fixed and added to bugfix-2.0.x

closed 2020-01-26 by boelle