MarlinFirmware/Marlin #16650: [BUG] (short description)

ro-ct, 2020-01-22:
Hey Guys,

I got two new Rumba32-Boards and I want to get it up and running with Marlin, but after flashing the firmware the board seems to hang in a reboot-loop.

### Setup
- Rumba32 1.0E
- no additional hardware is attached (no thermistor, no stepper drivers, no fans, no external power supply, nothing)
- Got everything up and running (https://github.com/Aus3D/RUMBA32/wiki/Getting-RUMBA32-Up-&-Running), device shows up as "STMicroelectronics Virtual COM Port"
- test with com-port successfully, counts endlessly (https://github.com/Aus3D/RUMBA32/wiki/COM-Port-Troubleshooting), flashed with Arduino 1.9.0

### My Configurations
- Marlin bugfix-2.0.x (as suggested from the Rumba32-Homepage), todays version
- configuration.h changes: 
   - #define SERIAL_PORT -1
   - #define MOTHERBOARD BOARD_RUMBA32
   - #define TEMP_SENSOR_0 998

### Behaviour
- device manager of windows 10 refreshes every 3..5 seconds, when board is attached over USB
- in Simplify3D, the board is in the same interval available/unavailable. When connecting with Simplify3D, the following happens:
```
Attempting connection at \\.\COM8… 
Testing plaintext communication protocol…
READ: ok
Connected to machine!
SENT: T0
READ: ok
SENT: M105
READ: ok T:0.00 /0.00 @:0
WARNING: Device unplugged while connected to port
Disconnected.
```
- on the one hand the automatic disconnection is one problem but also the temperature should report the `DUMMY_THERMISTOR_998_VALUE` (default 25) but not 0
 - response in Arduino serial monitor:
```
13:28:58.467 -> start
13:28:58.467 -> echo: Watchdog Reset
13:28:58.467 -> Marlin bugfix-2.0.x
13:28:58.467 -> 
13:28:58.467 -> echo: Last Updated: 2020-01-21 | Author: (none, default config)
13:28:58.467 -> echo:Compiled: Jan 22 2020
13:28:58.467 -> echo: Free Memory: 123891  PlannerBufferBytes: 1280
13:28:58.467 -> echo:Hardcoded Default Settings Loaded
13:28:58.467 -> echo:  G21    ; Units in mm (mm)
13:28:58.467 -> 
13:28:58.467 -> echo:Filament settings: Disabled
13:28:58.467 -> echo:  M200 D3.00
13:28:58.467 -> echo:  M200 D0
13:28:58.467 -> echo:Steps per unit:
13:28:58.467 -> echo: M92 X80.00 Y80.00 Z4000.00 E500.00
13:28:58.467 -> echo:Maximum feedrates (units/s):
13:28:58.467 -> echo:  M203 X300.00 Y300.00 Z5.00 E25.00
13:28:58.467 -> echo:Maximum Acceleration (units/s2):
13:28:58.467 -> echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
13:28:58.467 -> echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
13:28:58.467 -> echo:  M204 P3000.00 R3000.00 T3000.00
13:28:58.467 -> echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
13:28:58.467 -> echo:  M205 B20000.00 S0.00 T0.00 J0.01
13:28:58.467 -> echo:Home offset:
13:28:58.467 -> echo:  M206 X0.00 Y0.00 Z0.00
13:28:58.467 -> echo:PID settings:
13:28:58.467 -> echo:  M301 P22.20 I1.08 D114.00
```
- I measured the "kill"-pin of the Rumba32-board (PC5-pin of the stm32-controller): the voltage drops in the same interval from 3.3 V to Gnd.

### Flashing Problems
At the beginning I could not even flash Marlin because the _file name or extension were too long_. I tried a lot like choosing short paths, editing the build.path in the Arduino-preferences, etc. but nothing helped. It seems that the lcd-sources can lead to quite long paths so I removed the folder _extension-ui_ in `\src\lcd`. After that, flashing was not a problem any more. I just mention this here in case, this might be important again.

### What I have tried so far
- different computers
- different hosts
- different usb-cables
- enabling` #define WATCHDOG_RESET_MANUAL `
- both Rumba32-boards
Nothing changed, the boards still reboot

I really don't know what else I can try so I would appreciate any help or hint to overcome this issue.

Best regards in advance,
ro-ct

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4097758/config.zip)

xC0000005, 2020-01-23:
If this builds using HAL_STM32, those timers aren't firing correctly, which means if the watchdog is enabled, it's probably rebooting when the temp ISR doesn't work. You can test by disabling the watchdog in config and seeing if it stops rebooting. (I don't have a fix, and am under deadline, but I do know the ISR isn't firing).

sjasonsmith, 2020-01-23:
> (I don't have a fix, and am under deadline, but I do know the ISR isn't firing).

Is the timer issue you mention specific to the Rumba32, or other boards as well?

xC0000005, 2020-01-23:
I’ve seen it on two (Chitu and Malyan) and it repros when I debug on a bluepill. Are you seeing it on others?

sjasonsmith, 2020-01-23:
No, I have not, but I haven't been actively working with any STM32 boards for a while. Aren't the Chitu and Malyan boards supposed to be using the Maple-based STM32F1 HAL?

xC0000005, 2020-01-23:
Malyan V2 and M300 is 070 based, so…no STM32DUINO F1/F4 hal for them.

ro-ct, 2020-01-23:
> If this builds using HAL_STM32, those timers aren't firing correctly, which means if the watchdog is enabled, it's probably rebooting when the temp ISR doesn't work. You can test by disabling the [...]

Disabling the watchdog completely in configuration_adv works, the firmware now runs smoothly. I hope this get's fixed soon because I would feel better with the watchdog running like it should.
Thanks @xC0000005

xC0000005, 2020-01-23:
Disabling the watchdog is NOT a fix - it's a way to identify that yes, the watchdog is triggering because the temp ISR isn't running. Please don't run anything real without the watchdog.

boelle, 2020-01-26:
has anyone been able to reproduce this problem?

xC0000005, 2020-01-26:
Yes. It happens on a blue pill.

boelle, 2020-01-26:
marlin on a blue pill? thought i have seen it all but oh well :-D

xC0000005, 2020-01-27:
Malyan, Chitu, and Lerdge (the three boards I've been working most with)
aren't terribly friendly or helpful in terms of giving nice SWD connectors.
The bluepill does, and I've wired them up to RAMPS boards before. For this
particular problem, because it's something to do with the ISRs, there's no
need to have an actual printer attached - you can set a breakpoint on the
ISR and see it never fires (alternately, just put a bkpt block in the ISR
and watch it never get hit)

xC0000005, 2020-01-27:
I pulled the latest Arduino Core, and there's a fix for this issue -
https://github.com/stm32duino/Arduino_Core_STM32/issues/841
Without any other changes, the temp_isr is now firing on the bluepill, and I'll test the M200 soon. @ro-ct , you should probably pick up the latest ST core and see if it doesn't work for you.

boelle, 2020-01-27:
> I pulled the latest Arduino Core

would vs code not have done that anyway?

xC0000005, 2020-01-27:
PIO downloads releases that have been added to the manifests. On my computer i have my github repo for STM’s core symlinked so it takes the place of Arduino’s package. I pulled the most recent (not in a release that I know of yet) version and was pleased to find that the timers behave better now.

boelle, 2020-01-27:
but is it safe to say that this one if kind of confirmed and also what the fix is?

ro-ct, 2020-01-29:
> I pulled the latest Arduino Core, and there's a fix for this issue -
> stm32duino/Arduino_Core_STM32#841
> Without any other changes, the temp_isr is now firing on the bluepill, and I'll test the M200 soon. @ro-ct , you should probably pick up the latest ST core and see if it doesn't work for you.

I followed your advice and downloaded the newest version of the [Arduino STM Core](https://github.com/stm32duino/Arduino_Core_STM32) directly from github, following this [guide](https://github.com/stm32duino/wiki/wiki/Using-git-repository). **And it works**! The watchdog works, the steppers running well (had problems with that as well, probably because of the same timer-issue?) - so everything seems fine so far.
Before that I used the released core version 1.8.0, which I installed with Arduino IDE (board manager).

From my side, this topic can be closed.

Thank you for your help, much appreciated!

closed 2020-01-29 by boelle