MarlinFirmware/Marlin #16653: [BUG] PID Autotune stuck at 121ºC and triggering thermal runaway protection

Deses, 2020-01-22:
### Bug Description
I'm sending the code `M303 E0 S200 C8` through Pronterface on Windows 10 and after a while, it gets stuck at 121ºC and my Ender 3 printer starts beeping and it shuts down due to a thermal runaway warning.
If I set the temp to 200 ºC through the TFT screen or Pronterface, it reaches the desired temp without any issue, so I don't think it's really a thermal runaway issue.

I tested PID autotune with the Marlin firmware available from [BTT's github page](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2), a two months old build of Marlin 2.0.x and PID autotune it's working fine with that FW.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[DesesMarlin.zip](https://github.com/MarlinFirmware/Marlin/files/4100243/DesesMarlin.zip)

### Steps to Reproduce

1. Connect to my printer through Pronterface. Port COM4@115200 bauds.
2. Send the command `M303 E0 S200 C8`.
3. Turn off the printer from the power switch because it locks up and the beeping is deafening.

**Expected behavior:** PID Autotune manages to give me the data I want.

**Actual behavior:** Printer beeps and halts.

Deses, 2020-01-22:
I'm building a versión of the fw with PID_DEBUG enabled to see what's going on...

tatusah, 2020-01-22:
What environment do you have set in the platformio.ini? There has been some reports of using _USB (USB mass storage) environment versions having this about similar kind of problems with some users.

Deses, 2020-01-22:
I'm using `STM32F103RC_bigtree_512K`:
```

[env:STM32F103RC_bigtree_512K]
platform          = ststm32
framework         = arduino
board             = genericSTM32F103RC
board_upload.maximum_size=524288
platform_packages = tool-stm32duino
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
build_flags       = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py
  ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DUSE_USB_COMPOSITE -DHAVE_SW_SERIAL -DSS_TIMER=4 -DSTM32_FLASH_SIZE=512
build_unflags     = -std=gnu++11
lib_deps          = ${common.lib_deps}
  SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip
lib_ignore        = Adafruit NeoPixel, SPI
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
monitor_speed     = 115200
```
I'll take a look at this issue too, maybe they are related as this env profile also uses `USE_USB_COMPOSITE`.

Deses, 2020-01-22:
I'm using `STM32F103RC_bigtree_512K`:
```

[env:STM32F103RC_bigtree_512K]
platform          = ststm32
framework         = arduino
board             = genericSTM32F103RC
board_upload.maximum_size=524288
platform_packages = tool-stm32duino
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
build_flags       = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py
  ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DUSE_USB_COMPOSITE -DHAVE_SW_SERIAL -DSS_TIMER=4 -DSTM32_FLASH_SIZE=512
build_unflags     = -std=gnu++11
lib_deps          = ${common.lib_deps}
  SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip
lib_ignore        = Adafruit NeoPixel, SPI
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
monitor_speed     = 115200
```
Looks like the issue you linked is exaclty the same issue I'm having so I'll be closing this one.

Thank you for pointing me to it!! :)

closed 2020-01-22 by Deses

tatusah, 2020-01-22:
Ok, I would say it's the same issue then. It's just that your platformio.ini file is not a recent one as the use of USB_COMPOSITE and naming of the environments have been shuffled around some time ago. The current platformio.ini file has the USB_COMPOSITE in the _USB named environments. But good that you checked the the COMPOSITE flag in detail even when the environment naming was different.