MarlinFirmware/Marlin #16655: [FR] Less complex filament change

Liger0, 2020-01-23:
### Description

The filament change is irritating to me actually. This is what happens:
1. Change filament
2. I select to heat
3. Filament goes out and meanwhile I do other things
4. I insert the filament and hit to heat and meanwhile I do some other things because I don't like wasting minutes watching a nozzle heating
5. Temp cooled.
6. I hit to heat again.
7. Temp cooled again.
8. ....
100. ....
101. I now wait like a dumb the hotend heating all the time and insta-press load when the temp reaches the target because if I wait some seconds I am fucked again.
102. Yes, now it heats again because some seconds have passed.
103. FINALLY FILAMENT STARTS TO LOAD

### Feature Workflow

1. Change filament
2. Heating process
3. Filament goes out
4. Asks for new filament and if agreed it heats.
5. LOAD DIRECTLY THE FILAMENT WHEN HEATED

InsanityAutomation, 2020-01-23:
It's on my list to rework some of this and reduce waits but probably won't happen till after mrrf.

tfgcampos, 2020-05-14:
Another great option for filament change process would be to have an option to use the filament runout sensor as a trigger to start extruding the new filament, instead of having to click on the LCD knob. A delay could be added after runout sensor being triggered for the cases where the sensor is not near the extruder.

closed 2020-06-30 by github-actions[bot]