MarlinFirmware/Marlin #16661: Marlin 2.0.1 stuck at bootscreen when SD card removed

Thijsvanzwam, 2020-01-23:
When powering on my Anet A6 printer, running Marlin 2.0.1 on the Anet v1.7 mainboard with optiboot bootloader, the system halts at the bootscreen and with a running fan at 100%. (it powers up the fan shortly at startup but it gets stuck at that point.)
It only happens when the sd-card is removed. Starting up with a sd-card inserted, no problems at all.

boelle, 2020-01-24:
also remember to use bugfix 2.0.x as 2.0.1 is a frozen copy of 2.0.x

Thijsvanzwam, 2020-01-24:
here are my config files, sorry read all over it in the issue template.

I can replicate the bug each time by just powering on the printer with the sd card removed, when i reset the printer and put in the card, it powers on normally. When i insert the card at the moment it is stuck, it sometimes starts normal in a few second and sometimes it does not start up at all, remaining stuck at the bootscreen. Then i have to reset the printer and all is normal again. 

[config files.zip](https://github.com/MarlinFirmware/Marlin/files/4107132/config.files.zip)

houseofbugs, 2020-01-26:
Sounds like a bad SD card. Have you tried another to rule out issues with the card itself?

ManuelMcLure, 2020-01-26:
Uhhh... he reports that he sees a problem when he _doesn’t_ have an SD card inserted. I don’t see how a bad SD card would cause that.

Thijsvanzwam, 2020-01-28:
SD card is fine, i use several sd cards and the problem only occurs when there is no card inserted.

boelle, 2020-01-28:
my printer boots fine with no sd card inserted in the lcd
but then i have a re-arm with an sd card inserted so i have 2 options for sd card

@Thijsvanzwam did you try with bugfix 2.0.x or ony 2.0.1?

Thijsvanzwam, 2020-01-30:
tried only 2.0.1
Will try bug fix this week

boelle, 2020-01-30:
2.0.2 and bugfix are broken right now so hold off for maybe a week or 2

boelle, 2020-02-08:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-02-08 by boelle

thinkyhead, 2020-02-11:
And the SD detect pin is in good working order…? It is a mysterious one. If something at startup takes too long, Marlin will usually just restart rather than hang.

Thijsvanzwam, 2020-02-11:
tried 2.0.3, same issue.
back to 1.1.9 no problems at all

thinkyhead, 2020-02-11:
We can try staring at the code, but truly that never works. Can you try some experiments with various SD-related settings and see if something changes? Maybe you'll discover some weird interaction with another enabled feature that, when disabled, fixes the issue.