MarlinFirmware/Marlin #16663: [BUG] (short description)

Mosher23, 2020-01-24:
Hello,

i think i am experiencing some kind of communication bug on my SKR 1.4 Turbo. I can't or can but with a decent delay abort my prints.

My configuration:
Octopi (latest release) on Raspberry pi 4
SKR 1.4 turbo connected via UART to RPi
Marlin 2.0.1 bugfix

### Bug Description
1. Aborting print through Octopi works with great delay, like 30 sec

2. If i abort print on control screen of my Ender 3 Pro, printer aborts for a second and then continues printing. This happens when the print is sent through octopi.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/4107324/Configuration.files.zip)

### Steps to Reproduce

1. [First Step] Send your print through octopi to printer
2. [Second Step] Abort your print after it started printing, either on Octopi ( to reproduce 1 scenario) or on printer control display (2. scenario)

**Expected behavior:** [What you expect to happen]
Aborting the print immediately

**Actual behavior:** [What actually happens]
1. delay abort of the print, which can result in damaging of the printer parts
2. inability to abort the print 

#### Additional Information

If you can help i am ready to try out any solution

ellensp, 2020-01-24:
Try enabling EMERGENCY_PARSER in Configuration_adv.h  Otherwise the command queue has to process everything queued before it will abort.

Mosher23, 2020-01-24:
> Try enabling EMERGENCY_PARSER in Configuration_adv.h Otherwise the command queue has to process everything queued before it will abort.

thank you, this solved my 1. problem
2. is still there

thisiskeithb, 2020-01-24:
Enable `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT`. Those will tell OctoPrint to stop printing if you cancel via the printer’s LCD.

boelle, 2020-01-26:
@Mosher23 did the suggestion from @thisiskeithb solve the last problem?

Mosher23, 2020-01-26:
@boelle i had no opportunity to test it yet, but i will do tomorrow and post here results

boelle, 2020-01-26:
:-) lokking forward to it

boelle, 2020-01-28:
assuming this one solved and closing

closed 2020-01-28 by boelle

Mosher23, 2020-01-29:
Sorry for a late reply, but the solution has worked for me. Thank you