MarlinFirmware/Marlin #16664: [BUG] Probe vs Nozzle position limitation in Mesh Levelling

siteswapjuggler, 2020-01-24:
### Bug Description

In bed levelling (G29) the movement are limited to (X_MAX_POS - 10) instead of (X_MAX_POS) resulting in non symetrical mesh measurement if your prob offset make your pose to go further than this.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4108821/config.zip)

### Steps to Reproduce

1. use the configuration files provided (BED_SIZE 302, X_MAX_POS 314, X_PROBE_OFFSET -27)
2. G29 bed levelling
3. first probe point in the line will we 304 while it should be 309 (302 - 20 + 27)
4. last probe point in the line is 47 wich seems ok (20+27) despite I don't know where the 20 is coming from (MIN_PROBE_EDGE is set to 10)

**Expected behavior:**

Mesh probe point should limit the nozzle position to MAX_POS
The probe position should be limited by bed dimension and min probe egde (not the nozzle position)

**Actual behavior:** [What actually happens]

Mesh probe point limits the nozzle position to MAX_POS - 10 (MIN_PROBE_EDGE ?)

#### Additional Information

I haven't test if the problem occure with Y or with MIN_POS when probe is a positive number.
I tested

tatusah, 2020-01-24:
Based on the symptoms I suspect you are using probably Marlin 2.0.1 release? There is a problem with min probe edge doubling https://github.com/MarlinFirmware/Marlin/pull/16367 which is fixed in the bugfix-2.0.x branch. So try changing to the bugfix branch as it's newer than the 2.0.1 release and has the edge issue fixed.

siteswapjuggler, 2020-01-24:
Seem legit, I was on the 2.0.x but not on the more recent one. I'll check this as soon as possible (every printer are working right now).

boelle, 2020-01-30:
a lot of updates and fixing has happend in the last week, is the problem still there?

boelle, 2020-02-08:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-02-08 by boelle

siteswapjuggler, 2020-05-08:
Hello newer branch did indeed correct the problem tanks a lot for your listening.