MarlinFirmware/Marlin #16665: [BUG] Checksum Mismatch when printing

superdumnezeu, 2020-01-24:
### Bug Description
When printing with the MKS TFT28 connected to the SKR 1.3, the Checksum Mismatch error occurs, and the print stops for a few seconds. 
The same issue occurs when trying to use baby stepping
[image]

### My Configurations
SKR 1.3 with TMC2208 in UART mode
Stock TFT2.8 from Sidewinder X1 
Sidewinder X1 printer
Config used:
[Checksum Error.zip](https://github.com/MarlinFirmware/Marlin/files/4108874/Checksum.Error.zip)

### Steps to Reproduce
1. Start a print
2. Try to use baby stepping or wait for the print to stop 
3. The print stops for a few seconds and the Checksum Mismatch error message appears on the screen. 
4. The print resumes only to stop again in a few seconds

#### Additional Information
MKS TFT28 screen is connected in the AUX-1 port, in the top 4 pins. 
Printing from the USB or SD card generates the same issue.
I tried with 115200 and 250000 baud rate - it makes no difference. 
The original MKS Gen L board works without any issues, with the same configuration settings.

Necropaw, 2020-01-24:
The TFT2.8 on the Sidewinder X1 is a clone (as is the board). 
This is a hardware issue between SKR boards and the TFT2.8 clone.

ellensp, 2020-01-24:
my guess would be electrical interference...  MKS Gen L uses 0-5v serial signaling, SKR uses 0-3.3v so is more susceptible to electrical noise...

superdumnezeu, 2020-01-24:
> my guess would be electrical interference... MKS Gen L uses 0-5v serial signaling, SKR uses 0-3.3v so is more susceptible to electrical noise...

Forgot to mention i tried with an insulated wire, with the same results :(

boelle, 2020-01-28:
remember to only connect the "screen" in the wire at receiver end, if you connect at both ends you end up with ground loops

superdumnezeu, 2020-01-28:
> remember to only connect the "screen" in the wire at receiver end, if you connect at both ends you end up with ground loops

Can you please explain this in detail ? The screen is connected with a 4 wire cable from the TFT to the board. There are also 2 wires (red+black) that are used for power afaik.
[image]

boelle, 2020-01-28:
"screen" as in wire that has insulation and then a metal mesh arround it and then insualtion again

the metal mesh protects from noise

boelle, 2020-01-28:
so in your case you mean wire with plastic arround it, that you should always use so i was thinking your meant wire with the metal mesh in in

but yes your setup has loads of options for electrical noise to get inside the wires

closed 2020-04-29 by thinkyhead