MarlinFirmware/Marlin #16666: [FR] Teensy 4.0 Support

limitz, 2020-01-24:
Hey!

I thought it might be nice to try and get the new Teensy 4.0 to run Marlin. I've already started a branch, this is a shoutout for collaboration if anyone is thinking about doing the same?

I started with the the teensy35_36 HAL and worked my way up from there. It's still in early stages (it builds but I need to make a test setup). Any advice, help or peer review is appreciated. 

I have a branch here: https://github.com/limitz/Marlin/tree/teensy40

teemuatlut, 2020-01-27:
You should start your work from the bugfix branch and not the release.

limitz, 2020-02-02:
Thanks, just did.

kursatu, 2020-03-23:
@limitz What is the status of the teensy 4.0 branch?
I see some work being done, I may contribute, waiting on a ramps 1.5 to 
arrive so that I can clone and try your branch.
What is the board you use to with teensy 4.0, is it some ramps board and 
a shield adapter?
Given that teensy 4.0 is so powerful and has all the pins/hw/timers/interrupts ready 
to use , it would be quite fun to build Marlin on it.

limitz, 2020-03-25:
@kursatu I was just breadboarding it with some external mosfets and tmc5160s, don't think you need much more than that. However, I got lost in some preprocessor hell because several dependencies of marlin don't use the marlin HAL (obviously) and the arduino port for teensy didn't work for them (u8g2 for instance). Also tried to fix some PROGMEM stuff instead of just defining it as NOP, but that just opens another can of worms with bugs introduced by the fact that a lot of HALs "fix" pgm compilation errors by just defining it empty (serial writes expecting progmem string called with strings allocated on heap or stack.
In the meantime I got an ultratronic board and put this project on the backburner, because I think that the way marlin is structured now is flawed and would rather spend time on fixing those flaws than working around those flaws to get the teensy working :)

Hedda, 2020-05-12:
FYI, just submitted a feature request #17973 for Teensy 4.1 board support in Marlin

Teensy 4.1 was just released today and compared to Teensy 4.0 the biggest hardware news with Teensy 4.1 is a MicroSD-card slot, more I/O support, and integrated 10/100 Mbit Ethernet PHY (physical layer) support

https://www.cnx-software.com/2020/05/12/teensy-4-1-board-gets-longer-for-ethernet-microsd-slot-and-more-gpios/

About Teensy 4.1's new features:
https://pjrc.com/teensy-4-1-released/

Teensy 4.1 product page:
https://pjrc.com/store/teensy41.html

Teensy 4.1 Arduino compatible board is powered by the same NXP i.MX RT1062 Cortex-M7 crossover processor clocked at 600 MHz as Teensy 4.0, however Teensy 4.1 is about doubling in length in order to add a 10/100 Mbit Ethernet PHY, a MicroSD card slot, and offer more I/Os.

Teensy 4.1 specifications:

- SoC – NXP i.MX RT1062 Arm Cortex-M7processor at 600 MHz with  1024KB RAM (512KB is tightly coupled),
- Storage – 8 MB serial flash (64KB reserved for recovery & EEPROM emulation), MicroSD Socket, footprints for 2x extra QSPI chips such as flash or 8MB PSRAM chip
- USB – 1x micro USB port for power and programming
- Networking – 6-pin Ethernet header via 10/100 Mbit DP83825 PHY
- Expansion via through-holes and pads
  - USB – 2x USB ports, both 480 MBit/sec (5 pins with power management)
  - Storage – 1x SDIO (4 bit) native SD
  - Audio – 2x I2S Digital Audio, 1x S/PDIF Digital Audio
  - Serial – 8x Serial, all with 4 byte FIFO; 3x CAN Bus (1x with CAN FD)
  - 35x PWM pins
  - 40x digital pins, all interrupt capable
  - 18x analog pins, 2x ADCs on chip
  - 3x SPI, all with 16 word FIFO; 3x I2C, all with 4 byte FIFO
  - Total I/O pins – 55 (vs 40 for Teensy 4.0)
- Security – Cryptographic Acceleration; Random Number Generator
- Misc – RTC for date/time, 32x general-purpose DMA channels, Programmable FlexIO, Pixel Processing Pipeline, Peripheral cross triggering, Power On/Off management
- Power Supply – 5V via micro USB port
- Dimensions – 7.2 x 1.8 cm (TBC)

The software is mostly software compatible with Teensy 4.0, and can also be programmed in Arduino IDE using Teensyduino software add-on.

Hedda, 2020-06-24:
@limitz and @kursatu  What is the status of your Teensy 4.0 branches?

CrazzyFrenchDude, 2020-08-13:
Hello to everyone,  I am Crazzy French Dude, from France! Nice to be here! Actually developing a BoB named "PeeNaPle" for Teensy 4.1 with the help of some from the  Open Pnp community. Not much of a difference with 3D printers, cartesian machines! This board is modular and comes with lots of extension boards and adaptors so you can use your favorite stepper motor, power switch, relay etc with it! I am actually seeking serious developers to try to port Marlin to it and any supported Gerber controlling firmware! This project will be Open Hardware and Open Software, If any of you is interested feel free to ask any further information! Thanks much! CFD
You can have a glance at the pict below
[image]

Hedda, 2020-08-13:
@CrazzyFrenchDude That board design looks like it has some great potential! Please be sure to also post in #17973 too -> https://github.com/MarlinFirmware/Marlin/issues/17973

Since integrated ports for stepper-motors and MOSFETs directly on that board it will need something like the [RAMPS_1.4.4](https://reprap.org/wiki/RAMPS_1.4.4) or the upcoming [RAMPS 1.7](https://reprap.org/wiki/RAMPS_1.7) (which is the [latest stepper motor shield for 32-bit boards being worked on by MrAlvin](https://github.com/MrAlvin/RAMPS_1.7) and the RepRap community), right?

Btw, Biqu also makes as sell popular closed sourced branded RAMPS shields in the form of BigTreeTech [RAMPS 1.6 ](https://reprap.org/wiki/RAMPS_1.6) and [RAMPS_1.5](https://reprap.org/wiki/RAMPS_1.5)

One idea with that comes to mind is that IMHO you should really think about for the 3D-printing community with your board is that it could be awesome to aim at making an open hardware solution that can compete with the most popular closed sourced 3D-printer hardware boards, like the newest [BTT SKR V1.4](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3) (all-in-one board with plug-in stepper motors) or [BigTreeTech SKR Mini E3 V2.0](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3) (all-in-one board with embedded stepper motors) boards by Biqu. 

Wondering if you are planning on manufacturing small batches and selling boards via sites like Tindie, Crowd Supply, or similar?

CrazzyFrenchDude, 2020-08-13:
> @CrazzyFrenchDude That board design looks like it has some great potential! Please be sure to also post in #17973 too -> #17973
> 
> Since integrated ports for stepper-motors and MOSFETs directly on that board it will need something like the [RAMPS_1.4.4](https://reprap.org/wiki/RAMPS_1.4.4) or the upcoming [RAMPS 1.7](https://reprap.org/wiki/RAMPS_1.7) (which is the [latest stepper motor shield for 32-bit boards being worked on by MrAlvin](https://github.com/MrAlvin/RAMPS_1.7) and the RepRap community), same has has to be done with the [Re-ARM] controller board (https://reprap.org/wiki/Re-ARM) (which is an open-source hardware 32-bit board by the RepRap community). By the way, Biqu also makes as sell popular closed sourced BigTreeTech branded RAMPS shields in the form of [RAMPS 1.6 ](https://reprap.org/wiki/RAMPS_1.6) and [RAMPS_1.5](https://reprap.org/wiki/RAMPS_1.5)
> 
> Are you planning on getting any RAMPS yourself to see if can fit pins designed so could work as a plugin shield without wires? For example, checkout how RAMPS_1.4.4 is fitted to plug-into the [Re-ARM](https://reprap.org/wiki/Re-ARM) board here -> https://reprap.org/wiki/File:RAMPS144-02.JPG
> 
> Pinout for use with RAMPS 1.4 /RAMPS_1.4.4 is found here -> https://reprap.org/wiki/Re-ARM#Pinout_when_used_with_RAMPS1.4
> 
> One idea with that comes to mind is that IMHO you should really think about for the 3D-printing community with your board is that it could be awesome to aim at making an open hardware solution that can compete with the most popular closed sourced 3D-printer hardware boards, like the newest [BTT SKR V1.4](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3) (all-in-one board with plug-in stepper motors) or [BigTreeTech SKR Mini E3 V2.0](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3) (all-in-one board with embedded stepper motors) boards by Biqu.
> 
> Wondering if you are planning on manufacturing small batches and selling boards via sites like Tindie, Crowd Supply, or similar?

Hello Heda and thanks much for this warm welcome of the PeeNaPle bob! As a matter of fact, this board do not have any stepper drivers or power Mosfet directly on it! This is what makes it modular! Let's say you have a set of you favourite TMC2208 we got you covered! Just plug your driver on the daughter board, set your jumpers and you good to go, same as if you would with a BTT. If you need more steppers, just plug in the extension board and off you go, you need Wifi, sure, use the UEXT port. This board makes it easy for you build your controller with exactly and only what you need, it has an eco system ready extension package for extended! Actually in testing phase I am. planning for mass production so It will the coast at a very low entry point what would get the makers interest tick! The only thing I haven't done yet is work on any Firmware. But as you suggested it is good for Open Pnp and for 3D printing as well!

limitz, 2020-08-14:
> @limitz and @kursatu What is the status of your Teensy 4.0 branches?

I have since switched to ESP32 as my MCU of choice and am not planning to proceed with any Teensy related development.

CrazzyFrenchDude, 2020-08-15:
> > @limitz and @kursatu What is the status of your Teensy 4.0 branches?
> 
> I have since switched to ESP32 as my MCU of choice and am not planning to proceed with any Teensy related development.

ok then that's a pity

Gamester17, 2020-10-09:
FYI, it looks like Teensy 4.1 (and Teensy 4.0) support has now been merged via https://github.com/MarlinFirmware/Marlin/pull/19311 via PR from @bilsef

Marlin Firmware bugfix-2.0.x branch README.md now specifically list these as supported controller boards:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/README.md

**Teensy 4.0 / 4.1**
boards | processor | speed | flash | sram | logic | fpu
-- | -- | -- | -- | -- | -- | --
Teensy 4.0 | IMXRT1062DVL6A ARM-Cortex M7 | 600MHz | 1M | 2M | 3.3V | yes
Teensy 4.1 | IMXRT1062DVJ6A ARM-Cortex M7 | 600MHz | 1M | 2M | 3.3V | yes

Teensy 4.x is also supported in MarlinFirmware stable release branch but the description in the readme file is not as good:

https://github.com/MarlinFirmware/Marlin/blob/2.0.x/README.md

closed 2020-10-09 by ellensp

Gamester17, 2020-10-15:
Another FYI, just read about newly announced "iMX RT1064 uCOM" board from Embedded Artists so submitted a feature request:

https://github.com/MarlinFirmware/Marlin/issues/19749

Maybe depending on its price when general availability it might be an option for those looking at developing BoBs for Teensy 4.1