MarlinFirmware/Marlin #16668: [BUG] (M20 returns empty despite files present on storage medium)

rlayne, 2020-01-25:
### Bug Description

Issuing the M20 command is expected to return a list of files available from a mounted SD card on the controller; in this issue it only ever returns an empty list regardless of what is present on the SD card.

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4111540/Marlin.zip) 2.0.1 config
(problem persists, initially observed with Marlin 1.1.4)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Connect to printer as usual using repetier or any software with a SD card support or serial console
2. After initialisation, issue an M20 command or try and populate the file list

**Expected behavior:** Marlin responds with a list of files present on the root of the SD Card

**Actual behavior:** Marlin responds with an entirely empty list, i.e. nothing between begin/end

#### Additional Information

[image]

Initially encountered on Marlin 1.1.4 
Upgrading to Marlin 2.0.1 did not affect the issue
Altering firmware config has not affected the  issue:
1. halving SPI rate
2. turning off/on long filename support
3. turning off/on binary transfer support
4. Correct file list is visible on the controller LCD itself and can be printed.

thisiskeithb, 2020-01-25:
Can you try with the latest `bugfix-2.0.x`? I'm running the latest on several machines and `M20` output is as expected:
```
Send: M20
Recv: Begin file list
Recv: MK3_BT~1.GCO 45855
Recv: End file list
Recv: ok
```

rlayne, 2020-01-25:
> Can you try with the latest `bugfix-2.0.x`? I'm running the latest on several machines and `M20` output is as expected:
> 

Yes certainly. Assuming the configs are the same between 2.0.1 and x I'll be able to do that over the next day or so.

AnHardt, 2020-01-25:
Are there any files with the extensions *.g, *.gco. *.gcode on the card?

Marlin does not display endings like: *.exe, *.bin, *.hex, *txt, *.log, ...
just files it expects to contain g-code.

rlayne, 2020-01-25:
> Are there any files with the extensions *.g, *.gco. *.gcode on the card?

Yes, there are always files of all three extensions on the card for testing purposes, I've also tried removing none gcode files or not, as a further test, and removing folders and just using the root of the card, too.

sjasonsmith, 2020-01-26:
I tried reproducing the problem using your exact configuration with 2.0.1. I had no issues with M20 or M21.

My testing was all done with a generic serial terminal (TeraTerm), as I don't have Repetier or any other host software installed on my machine.

I tested using a Mega2560 with RAMPS 1.4. I tested first with a RepRapDiscount Full Graphics display, then again with a RepRapDiscount character display so that I could use your configs unaltered.

I tested with both cheap 128 MB Creality SD cards, as well as a quality 16 GB SanDisk card.

I even tried using the SD card from a Mac so that it would get filled up with garbage hidden files/folders, which I've seen cause issues on some other (non-Marlin) machines. It all still worked fine.

I wish I had better news for you, but I just can't observe any issues on my end.

rlayne, 2020-01-26:
> I wish I had better news for you, but I just can't observe any issues on my end.

Thanks a lot for going through the trouble of doing that for me.  I was afraid this would be the case.  The firmware version on my MKS TFT32 v4 is higher than the most recent available stock firmware.  There is an FLSun splash screen on boot but other than that I don't know what else has been messed with, but whatever it is may be responsible for the error.  

I'm not sure exactly how the M20 command is relayed to the MKS TFT32 board, from Marlin but assuming I can find some documentation on how it handles messaging I might try talking to it directly via GPIO...  or for the sake of £30 I could just buy a replacement immediately which will likely just work.

My main motivation for wanting the sd card to operate correctly is when I print via USB, there's a chance the print will stall (print head stops, refuses further commands, marlin spams waiting for user input, ), which I haven't figured out is due to Cura temperature request spam or some USB deficiency.

sjasonsmith, 2020-01-26:
How does the TFT display fit into this? From your configs, this looked like a simple RAMPS board with a RepRapDiscount character display attached, with your RAMPS connected directly to the host PC over USB.

rlayne, 2020-01-27:
Ah, so then to be clear, the configs were largely copied from an original
1.1.4 distro with my printer, which also comes with a MKS TFT32 v4 where
this board provides the SD CARD.  It's entirely possible something is
misconfigured if this isn't clear from the config.

boelle, 2020-01-28:
will close this one as its mostly a config problem
config files say its a RepRapDiscount character display but a TFT is mentioned

closed 2020-01-28 by boelle