MarlinFirmware/Marlin #16670: [FR] Custom config file

guestisp, 2020-01-25:
Please add a custom config file for user settings.
Every time we merge Marling from vanilla to our custom version, we have to do A LOT of manual merge due to changes in Configuration.h and Configuration_adv.h

Please add 1 or 2 custom files that will never change in vanilla, something like `Configuration.user.h` and `Configuration_adv.user.h`

In this way , users could change their own settings (like offsets and so on) in `*.user.h` files and at the same time you can safely update the stock configuration.

limitz, 2020-01-25:
I agree, even more because I have multiple configurations (testing, production). The problem is not only for the config, but also if you want to alter platform.ini if you are testing with new boards. 

I usually do it by having different branches, but I made a PR for a possible solution if you're using PlatformIO.

thinkyhead, 2020-01-26:
We're working on the overall problem in a separate repository. There will be various options to try out of this effort, and things that work best will be rolled out with 2.1 and later versions. But consider, there is good reason for keeping configurations in two files that are standardized with each tagged version, and we are careful to track and provide sanity checks for the changes from version to version. All this data can be collated and used in conjunction with the configurator project developing over at https://github.com/MarlinFirmware/AutoBuildMarlin/tree/configurator

closed 2020-01-26 by thinkyhead