MarlinFirmware/Marlin #16671: [BUG] Dual Z provides incorrect current/stealthchop modes

CarlitoGrey, 2020-01-25:
### Bug Description

Enabling Dual Z causes startup current to differ, and stealthchop to be disabled. 

```
Recv: 		X	Y	Z	Z2	E	E1
Recv: Enabled		false	false	false	false	false	false
Recv: Set current	730	730	850	800	730	730
Recv: RMS current	703	703	826	795	703	703
Recv: MAX current	991	991	1165	1121	991	991
Recv: Run current	22/31	22/31	26/31	25/31	22/31	22/31
Recv: Hold current	11/31	11/31	13/31	12/31	11/31	11/31
Recv: CS actual	11/31	11/31	13/31	12/31	11/31	11/31
Recv: PWM scale	13	13	15	14	13	13
```

### My Configurations

Marlin 2.0.x branch [My code is here](https://github.com/CarlitoGrey/Marlin)
SKR 1.1 Pro
TMC 2208's in UART
Pins is set to default as of 2.0.1, although I did try adding: 
`  #define Z2_SERIAL_TX_PIN PD6
  #define Z2_SERIAL_RX_PIN PD6`

`#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2208
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2208
#define E1_DRIVER_TYPE TMC2208
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988`

`  #if AXIS_IS_TMC(Z)
    #define Z_CURRENT       800
    #define Z_CURRENT_HOME  Z_CURRENT
    #define Z_MICROSTEPS     16
    #define Z_RSENSE          0.11
    #define Z_CHAIN_POS      -1
  #endif

  #if AXIS_IS_TMC(Z2)
    #define Z2_CURRENT      800
    #define Z2_CURRENT_HOME Z2_CURRENT
    #define Z2_MICROSTEPS    16
    #define Z2_RSENSE         0.11
    #define Z2_CHAIN_POS     -1
  #endif`

### Steps to Reproduce

#define Z2_DRIVER_TYPE TMC2208
#define Z_DUAL_STEPPER_DRIVERS

**Expected behavior:** [What you expect to happen]

Current and stealthchop settings should match for Z and Z2

**Actual behavior:** [What actually happens]

Current and stealthchop settings differ.

boelle, 2020-01-28:
work fine for me on bugfix 2.0.x which is newer than 2.0.1

[...]
Send: M122
Recv: 		X	Y	Z	Z2	E
Recv: Address		0	1	2	0	3
Recv: Enabled		false	false	false	false	false
Recv: Set current	800	800	800	800	800
Recv: RMS current	795	795	795	795	795
Recv: MAX current	1121	1121	1121	1121	1121
Recv: Run current	25/31	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31	12/31
Recv: CS actual	12/31	12/31	12/31	12/31	12/31
Recv: PWM scale	16	15	14	17	27
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: tstep		max	max	max	max	max
Recv: pwm
Recv: threshold	49	49	329	329	30
Recv: [mm/s]		100	100	3	3	30
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time	4	4	4	4	4
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2	2
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: stst		*	*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0C:00:00
Recv: 		Y	0xC0:0C:00:00
Recv: 		Z	0xC0:0C:00:00
Recv: 		Z2	0xC0:0C:00:00
Recv: 		E	0xC0:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: ok
[...]

boelle, 2020-01-28:
remember to do M502 and then M500 after you flash firmware, i made that mistake and my microstep did not match

CarlitoGrey, 2020-01-28:
@boelle thank you, this was my mistake.

closed 2020-01-28 by CarlitoGrey