MarlinFirmware/Marlin #16672: SDIO read errors

KaSroka, 2020-01-25:
I'm using bugfix-2.0.x (bd550bb45) on my Sapphire Pro (MKS Robin Nano). The SD card on this board is connected using SDIO. I'm observing the "random wandering" of my print head as in issue #5772 and #5806 but they are on SPI, not SDIO.  Generally, the printer randomly tries to go to near 0 location of random axis due to not reading full line correctly.

Typically this can be solved by a Marlin configuration: `SD_CHECK_AND_RETRY` but it doesn't affect SDIO. That's ok as SDIO has its own retry mechanism. It just seems that it's not working as it should or I'm exceeding the retry count for a correct read.

I've tried two different SD cards, without any luck. I've tried moving cables away from traces between SD card slot and CPU, without any luck. I've tried to print via USB - it worked without any problem.

Right now it's impossible for me to finish any print longer than 6h (random wandering in X or Y is not a big issue, but Z ends up badly crashing into the print) and using USB is a no go in my current setup.

boelle, 2020-01-25:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

KaSroka, 2020-01-25:
Zipped configuration: 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4112146/Configuration.zip)

jmz52, 2020-01-25:
There is a hard-coded retry counter for SDIO read operations
Marlin\src\HAL\HAL_STM32F1\sdio.cpp:
```
bool SDIO_ReadBlock(uint32_t blockAddress, uint8_t *data) {
  uint32_t retries = 3;
  while (retries--) if (SDIO_ReadBlock_DMA(blockAddress, data)) return true;
  return false;
}
```
I personally had no problems with SDIO ever since this retry cycle was added (https://github.com/MarlinFirmware/Marlin/pull/13506), but my tests were less than 2 hours long.
You can try to increase retries counter to check if this is what causes your prints to stop.

boelle, 2020-01-28:
@KaSroka did you try what @jmz52 suggested?

atomphil, 2020-01-29:
My printer struggle with the same mistake. Please share if there is a solution.

atomphil, 2020-02-03:
I reinstalled Twotrees Firmware 1.0.3 on my Sapphire Pro just for testing and it is printing without errors for three hours now. So something didn´t work well in Marlin. It is a shame, that MKS didn´t publish the sources!

Bob-the-Kuhn, 2020-02-04:
A couple of other things that you might try:
- comment out line 81  ` sdio_set_clock(SDIO_CLOCK);` - this increases the clock from ??? KHz to 24MHz.  Maybe the slower clock rate will do the trick.
- comment out line 80  ` sdio_set_dbus_width(SDIO_CLKCR_WIDBUS_4BIT);` - this takes it from a one bit bus to a four bit bus.  Maybe there's noise on one of the added lines.

GeraltR, 2020-02-06:
@Bob-the-Kuhn I try you idea but isn't changes. I is still the same.

Bob-the-Kuhn, 2020-02-06:
Prints OK via USB but SD card print of the same thing fails after 6 hours.

**after 6 hours** implies a thermal problem or a counter overflowing.

Any way to provide additional cooling to the controller board?  I see there are slots on the bottom lid.  Maybe a fan directed there?

GeraltR, 2020-02-06:
My first thought was: reason must by more temperature on main board. I was opening controller board and addet aditional funs. It was still the same error. I think problem is from  Marlin functions read-out from SD card.

Bob-the-Kuhn, 2020-02-07:
Sorry - I'm out of ideas.

jmz52, 2020-02-07:
If you are using DC powered hotbed try to run a print on cold bed to verify that this problem is not caused by DC PSU

atomphil, 2020-02-07:
> A couple of other things that you might try:
> 
>     * comment out line 80  ` sdio_set_dbus_width(SDIO_CLKCR_WIDBUS_4BIT);` - this takes it from a one bit bus to a four bit bus.  Maybe there's noise on one of the added lines.

This dosn´t work for me. The SD-Card is not readable after that.

>     * comment out line 81  ` sdio_set_clock(SDIO_CLOCK);` - this increases the clock from ??? KHz to 24MHz.  Maybe the slower clock rate will do the trick.

This has significantly improved the printing result, reading errors seem to occur much less frequently. But every now and then strange behavior occurs.

GeraltR, 2020-02-09:
@jmz52 thank you form idea, but error was still. 
Yesterday I uncommented SPI_SPEED SPI_QUARTER_SPEED (I'v  already set SPI_SPEED SPI_HALF_SPEED it worked too) but it is small step to succes. Key to good working is value SDIO_CLOCK in sdio.h. I tryed various value and 2700000 (27kHz) is the best of them. Various was higher of her. My printer working thre task: 2 x 5h and 3h and few
 in 1,5h and wasn't error.
[image]

atomphil, 2020-03-03:
MKS have finally published their source code: https://github.com/makerbase-mks/MKS-Robin-Nano

25mmHg, 2020-03-03:
hope! have the same Problem with my Marlin 2.0.4.4 at sapphire PRO (MKS Robin nano)

KaSroka, 2020-03-08:
Sorry for late response. I think that all HW related issues (like more cooling or PSU problem) are not related. I was able to finish a lot of very long prints (up to 26h) without any issue using OctoPrint. I think that the problem comes down to two possibilities:
- HW problem on MKS Robin Nano (maybe the SD traces are near some noisy lines).
- SW problem in SDIO library.

As @GeraltR showed, lowering of SDIO clock helps - this may idnicate HW problem or wrong SDIO init.

For me - switching to OctoPrint fixes all the issues.

@atomphil  it would be nice to upmerge their changes to Marlin (like support for 3.5" LCD)

GeraltR, 2020-03-08:
I tryed to confirmation two the same file GCode first I used Marlin (load from SD card) and next i oryginal firmware. In Marlin this problem return every time but not the same file place. In oryginal firmware never was problem.
Similar problem i had in octoprint when I printed with speed 115200. I changed speed to 19200 and problem don't exists.
p.s. A long, long time ago I had internet with 19200 this is hight speed.

Malderin, 2020-03-13:
the same problem with my Marlin 2.0.4.4 at MKS Robin nano

Miraculix200, 2020-03-13:
In my case (with MKS Robin Nano 1.2) the problem disappeared when changing this line in Marlin/src/HAL/HAL_STM32F1/sdio.h:

#define SDIO_CLOCK                           18000000 /* 18 MHz */
to
#define SDIO_CLOCK                           4500000 /* 4.5 MHz */

atomphil, 2020-04-02:
I fixed the hardware of my MKS Robin Nano 1.1 by soldering an additional capacitor (0603 4.7µF) in parallel on top of C28 / C29. That solves the problem for me, no mystery moves anymore even when fast printing.

To be on the safe side, I now also set the SDIO_CLOCK to 4.5 MHz and the read retry to 16, as 25mmHg suggests.

```
Marlin/src/HAL/HAL_STM32F1/sdio.h

#define SDIO_CLOCK 4500000 /* 4,5 MHz */
```

```
Marlin/src/HAL/HAL_STM32F1/sdio.cpp

bool SDIO_ReadBlock(uint32_t blockAddress, uint8_t *data) {
  uint32_t retries = 16; 
```

boelle, 2020-06-20:
@KaSroka is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle