MarlinFirmware/Marlin #16673: [BUG] z axis noise on SKR mini E3

JohnGrab, 2020-01-25:
## Bug Description

I tried to update my printer to Marlin 2.0.1.
Since this update the z axis is making loud noises.
In the latest Bugfix Version the problem is still present.

My setup:

Printer:              Ender 3
Mainboard:       BIGTREETECH-SKR-mini-E3
Equipment:       BLTouch

## Further Analysis

I switched the Z and Y axis and the same loud noise is coming for the Y movement and the Z does not make any noise so the steppers are fine.

Shown in the videos switchedZandYconnectionsPart1.mp4 & switchedZandYconnectionsPart2.mp4

### My Configurations

platformio.ini, Configuration.h and Configuration_adv.h included in zip
also included is the firmware.bin

### Steps to Reproduce

1. Boot the latest Bugfix Version
2. move z axis or
3. perform auto home

Shown in the video movementOfZNois.mp4

**Expected behavior:** quit movement of the z axis
**Actual behavior:** loud and scratching sound on z axis movement

#### Additional Information

[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/4112195/MarlinConfig.zip)
[ZAxisNoiseVideos.zip](https://github.com/MarlinFirmware/Marlin/files/4112197/ZAxisNoiseVideos.zip)

JohnGrab, 2020-01-25:
If it helps i can analyse the voltage differences between the Y and the Z axis

Caj109, 2020-01-25:
Ive Just checked your Config_adv.

You do have "Hybrid_Threshold" enabled which could be a cause of the loud noise. so basically the steppers are switching cycle for higher torque requirements. you should be able to disable this in vscode and recompile.

sjasonsmith, 2020-01-25:
> You do have "Hybrid_Threshold" enabled

This is still the default in the SKR Mini E3 examples in the Config repo's master/2.0.1 branches. It is disabled in the _base_ branch, but I'm not sure what the process is of rolling those changes into the other branches.

JohnGrab, 2020-01-25:
I just recompiled and the loud noise is gone so  comment Hybrid_Threshold will fix the problem

Caj109, 2020-01-25:
> I just recompiled and the loud noise is gone so comment Hybrid_Threshold will fix the problem

Glad you got it fixed, it does catch people off guard. But suppose people prefer it so it doesnt miss steps etc

boelle, 2020-01-25:
closing since solved with a config change

closed 2020-01-25 by boelle