MarlinFirmware/Marlin #16674: [BUG] CoreXZ sensorless homing is enforced for z despite z probe enabled

organizedkaos, 2020-01-25:
### Bug Description

Using latest bugfix as of 25 Jan 2020, corexz kinematics with x axis sensorless homing enabled and z probe enabled is enforced to require z sensorless homing.

### My Configurations

The only changes made to reproduce the error was to download the bugfix repo, set the proper board (mks sgen L), set the tmc drivers (2209), set the fixed probe and set the sensorless homing for x and y.

### Steps to Reproduce

1. Download bugfix repo
2. Set board (mks sgen L)
3. Set tmc2209 drivers for x, y, z, and e0
4. Specify z min probe uses endstop and fixed probe (inductive probe)
5. Set sensorless homing for x and y axis
6. Compile fails with sanity check stating z axis requires sensorless homing for corexz kinematics due to x axis being set to sensorless homing

**Expected behavior:**

On corexz with x axis sensorless z homing should still be possible with endstop switch or probes (inductive, bltouch etc.).

**Actual behavior:**

Marlin fails to compile due to z axis sensorless homing enforcement on corexz.

#### Additional Information

To fix the situation I commented out the sanity check line regarding coreXZ and sensorless homing for x and z axis.

Marlin compiled and I was able to home all three axis in this configuration.  X axis homes with sensorless, y axis homes with sensorless, and z axis homes with inductive probe.

The only line changed to accomplish this beyond config and config_adv was the sanity check line.

Seems coreXZ should be able to choose z axis sensorless or z axis sensor (switch, probe etc.) regardless if x axis is set to sensorless.

I have tested this on my corexz build and as far as homing is concerned all three axis are working properly.

boelle, 2020-01-25:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

organizedkaos, 2020-01-26:
Here are the configs that are currently working for me.

When the sanity check line for corexz sensorless homing is commented out the build compiles and the printer works correctly with an inductive probe on z homing.

X and Y work on sensorless homing.

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4113108/Configs.zip)

boelle, 2020-01-30:
a lot of updates and fixing has happend in the last week, is the problem still there?

organizedkaos, 2020-02-03:
Yes corexz z sensorless is still enforced.  The same solution still applies ie comment the sanity check li e to compile.

boelle, 2020-02-08:
@organizedkaos if you have the skills then a PR is the fatest way to get it fixed

@thinkyhead does the proposed fix make any sense?

thinkyhead, 2020-02-11:
> does the proposed fix make any sense?

No, but this does:
```cpp
// Sensorless homing is required for both combined steppers in an H-bot
#if CORE_IS_XY && X_SENSORLESS != Y_SENSORLESS
  #error "CoreXY requires both X and Y to use sensorless homing if either one does."
#elif CORE_IS_XZ && X_SENSORLESS != Z_SENSORLESS && !HOMING_Z_WITH_PROBE
  #error "CoreXZ requires both X and Z to use sensorless homing if either one does."
#elif CORE_IS_YZ && Y_SENSORLESS != Z_SENSORLESS && !HOMING_Z_WITH_PROBE
  #error "CoreYZ requires both Y and Z to use sensorless homing if either one does."
#endif
```

commit e442663e (2020-02-11): No Z sensorless req'd if homing with probe

Fixes #16674
closed 2020-02-11 by thinkyhead