MarlinFirmware/Marlin #16678: G29 Sometimes "randomly" gives me ========= and not a value. Shifting Z down in area.

Magiske, 2020-01-26:
Some times when I print, the Z axis at one area of the bed moves down. I can see the Z screw turn aprox 10 degree. See picture. It's always on the back(upper) right corner.  And only on print that covers the area.

I am running 4x4 with ABL_BILINEAR_SUBDIVISON = 3.
Did two G29 right after the other. First one have the issue. But looks like it's only on the first run after the printer have been turned on. So quick fix = G29*2

SENDING:G29
Bilinear Leveling Grid:
      0      1      2      3
 0 -0.065 -0.135 -0.072 +0.135
 1 -0.127 -0.170 -0.033 +0.207
 2 -0.067 -0.085 +0.075 +0.345
 3 -0.095 +0.028 +0.270 +0.665
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 -0.06500 -0.09324 -0.12148 -0.13500 -0.12935 -0.10898 -0.07250 -0.01407 +0.06046  =======
 1 -0.09037 -0.11618 -0.14199 -0.15111 -0.13720 -0.10660 -0.06167 +0.00159 +0.07917  =======
 2 -0.11574 -0.13912 -0.16250 -0.16722 -0.14504 -0.10421 -0.05083 +0.01725 +0.09788  =======
 3 -0.12750 -0.14833 -0.16917 -0.17000 -0.14130 -0.09259 -0.03250 +0.03991 +0.12370  =======
 4 -0.11333 -0.13293 -0.15252 -0.15157 -0.12009 -0.06805 -0.00491 +0.06991 +0.15583  =======
 5 -0.08556 -0.10380 -0.12204 -0.11981 -0.08728 -0.03427 +0.03019 +0.10690 +0.19507  =======
 6 -0.06750 -0.07991 -0.09231 -0.08500 -0.04889 +0.00694 +0.07500 +0.15685 +0.25093 +0.34500
 7 -0.07019 -0.06932 -0.06845 -0.04954 -0.00505 +0.05748 +0.13352 +0.22606 +0.33210 +0.43815
 8 -0.08259 -0.06396 -0.04534 -0.01102 +0.04437 +0.11545 +0.20176 +0.30821 +0.42989 +0.55157
 9 -0.09500 -0.05861 -0.02222 +0.02750 +0.09380 +0.17343 +0.27000 +0.39037 +0.52769 +0.66500

SENDING:G29
Bilinear Leveling Grid:
      0      1      2      3
 0 -0.060 -0.135 -0.077 +0.122
 1 -0.135 -0.170 -0.035 +0.207
 2 -0.080 -0.067 +0.080 +0.342
 3 -0.082 +0.035 +0.275 +0.668
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 -0.06000 -0.08991 -0.11981 -0.13500 -0.13093 -0.11213 -0.07750 -0.02139 +0.05056 +0.12250
 1 -0.08981 -0.11593 -0.14205 -0.15176 -0.13891 -0.10964 -0.06602 -0.00393 +0.07253 +0.14898
 2 -0.11963 -0.14196 -0.16429 -0.16852 -0.14689 -0.10716 -0.05454 +0.01354 +0.09450 +0.17546
 3 -0.13500 -0.15296 -0.17093 -0.17000 -0.14157 -0.09426 -0.03500 +0.03787 +0.12269 +0.20750
 4 -0.12417 -0.13748 -0.15080 -0.14602 -0.11499 -0.06587 -0.00500 +0.06942 +0.15559 +0.24176
 5 -0.09889 -0.10698 -0.11508 -0.10676 -0.07512 -0.02707 +0.03306 +0.10784 +0.19471 +0.28157
 6 -0.08000 -0.08083 -0.08167 -0.06750 -0.03259 +0.01731 +0.08000 +0.15898 +0.25074 +0.34250
 7 -0.07657 -0.06694 -0.05731 -0.03333 +0.00993 +0.06754 +0.13907 +0.22902 +0.33289 +0.43676
 8 -0.07954 -0.05741 -0.03528 +0.ª00083 +0.05510 +0.12335 +0.20704 +0.31178 +0.43195 +0.55213
 9 -0.08250 -0.04787 -0.01324 +0.03500 +0.10028 +0.17917 +0.27500 +0.39454 +0.53102 +0.66750
[image]

Running Ender 3 with Fysetc Cheetah 1.1 and BLTouch.

boelle, 2020-01-26:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Magiske, 2020-01-27:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4115834/Configuration.zip)

Magiske, 2020-01-27:
[printer1.log.zip](https://github.com/MarlinFirmware/Marlin/files/4118150/printer1.log.zip)

Magiske, 2020-01-28:
Updated with that it can be reproduces 100% at first G29(after off/on)

Magiske, 2020-02-02:
It looks like when I G29 and get the "faulty" entries. I can desplay the current "faulty" entries. But if I store and recall the data. It's now vaild? (I have removed most of the data since it's the same for ease of readability!)

Bilinear Leveling Grid:
      0      1      2      3
 0 +0.815 +0.460 +0.223 +0.028
 1 +0.652 +0.320 +0.058 -0.148
 2 +0.588 +0.195 -0.062 -0.270
 3 +0.515 +0.142 -0.100 -0.260
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 +0.81500 +0.69231 +0.56963 +0.46000 +0.37056 +0.29417 +0.22250 +0.15435 +0.09093  =======
 9 +0.51500 +0.38602 +0.25704 +0.14250 +0.04898 -0.03009 -0.10000 -0.15944 -0.20972 -0.26000
M420 v

Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 +0.81500 +0.69231 +0.56963 +0.46000 +0.37056 +0.29417 +0.22250 +0.15435 +0.09093  =======
 9 +0.51500 +0.38602 +0.25704 +0.14250 +0.04898 -0.03009 -0.10000 -0.15944 -0.20972 -0.26000
M420 l
M420 v
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 +0.81500 +0.69231 +0.56963 +0.46000 +0.37056 +0.29417 +0.22250 +0.15435 +0.09093  =======
 9 +0.51500 +0.38602 +0.25704 +0.14250 +0.04898 -0.03009 -0.10000 -0.15944 -0.20972 -0.26000
M420 S1
M420 v
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 +0.81500 +0.69231 +0.56963 +0.46000 +0.37056 +0.29417 +0.22250 +0.15435 +0.09093  =======
 9 +0.51500 +0.38602 +0.25704 +0.14250 +0.04898 -0.03009 -0.10000 -0.15944 -0.20972 -0.26000
M500
M501
M420 v
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 +0.81500 +0.69231 +0.56963 +0.46000 +0.37056 +0.29417 +0.22250 +0.15435 +0.09093 +0.02750
 1 +0.75722 +0.63880 +0.52037 +0.41278 +0.32145 +0.24096 +0.16583 +0.09605 +0.03164 -0.03278
 2 +0.69944 +0.58528 +0.47111 +0.36556 +0.27235 +0.18775 +0.10917 +0.03775 -0.02765 -0.09306
 3 +0.65250 +0.53907 +0.42565 +0.32000 +0.22519 +0.13815 +0.05750 -0.01509 -0.08130 -0.14750
 4 +0.62389 +0.50426 +0.38462 +0.27454 +0.17832 +0.09165 +0.01111 -0.06239 -0.12976 -0.19713
 5 +0.60611 +0.47675 +0.34739 +0.23074 +0.13340 +0.04876 -0.03028 -0.10422 -0.17258 -0.24093
 6 +0.58750 +0.45167 +0.31583 +0.19500 +0.09731 +0.01463 -0.06250 -0.13537 -0.20269 -0.27000
 7 +0.56389 +0.42817 +0.29246 +0.17213 +0.07531 -0.00611 -0.08111 -0.15052 -0.21350 -0.27648
 8 +0.53944 +0.40710 +0.27475 +0.15731 +0.06215 -0.01810 -0.09056 -0.15498 -0.21161 -0.26824
 9 +0.51500 +0.38602 +0.25704 +0.14250 +0.04898 -0.03009 -0.10000 -0.15944 -0.20972 -0.26000

thinkyhead, 2020-02-02:
That’s interesting. A bad value is stored as a `NaN` float. This value poisons others when you add or subtract from it, and it shouldn’t convert into a numerical float even on load and store. I’ll poke around the code and see if there’s some obvious reason why this occurs.

Magiske, 2020-02-03:
The reading from the probing has always come back as valid. I presume the subdividing is what introduces the invalid "numbers"?

Magiske, 2020-02-15:
So, I believe I have figured out what is(was) wrong. 
Even doing a M502/G28/G29/M500/M501, it sometimes gave me an "EEPROM version mismatch (EEPROM=? Marlin=V74)". I notice that after upgrading to lates 2.0.3.bugfix build, after I now do a M520, I now newer get an error. And I can no longer reproduce the issue. So have something been changed handling the eeprom?

But even with a corrupt/version mismatch, a G29 should return valid entries. So could it be generating some corrupt values from  "undeclared" variables?

thinkyhead, 2020-02-15:
It's hard to say. Under normal conditions this is not seen, but I'll keep an eye out for EEPROM code that seems like it could be involved in corrupting things. The EEPROM code is due for an overhaul anyway.

closed 2020-02-15 by thinkyhead