MarlinFirmware/Marlin #16679: [BUG] Bigtreetech SKR e3 DIP 1.0 Z stepper configuration initially wrong

ss89, 2020-01-26:
### Bug Description
When compiling the STM32F103RE_bigtree env in visual studio platformio for the bigtreetech skr e3 dip 1.0 with 4 tmc2208 in uart mode the z-axis stepper in not initialized correctly.
This is a repeatable behaviour that i get on every startup of the board.
It is suddenly fixed, when i use the "Failsafe settings", however i did remove the eeprom file from the sd card, so there shouldn't be any issue with saved settings.
PlatformIO reports that it uses TMCStepper 0.6.1 (seems to be the one from teemuatlut) which depends on SoftwareSerialM.
Here is a link to my gitlab for checking out the changes that i did for my config:
https://git.x2d2.de/3D-Printing/Marlin-Firmware
Please see the recent-ender-3-bugfix-2.0.x-skr-e3-dip branch.
I also did try the current 2.0.1 first, because that's the version that i intended to upgrade to.

### My Configurations
https://git.x2d2.de/3D-Printing/Marlin-Firmware/-/archive/recent-ender-3-bugfix-2.0.x-skr-e3-dip/Marlin-Firmware-recent-ender-3-bugfix-2.0.x-skr-e3-dip.zip

### Steps to Reproduce

1. compile firmware with platformio
2. copy firmware.bin to sd card
3. insert sd card into printer and turn it on
4. wait until printer finished booting
5. connect via usb cable and execute M122
6. wrong microsteps indicate that the initialization is done wrong (there are some other settings wrong too)
7. use the displays menu to execute "Failsafe settings"
8. check M122 again and see that the microsteps are now correctly set

**Expected behavior:**
Initialization should be equal for all steppers.
Currently only Z seems to be different

**Actual behavior:**
Z stepper is setup wrong, microsetting set to 8 instead of 256

#### Additional Information
I did not have that issue with a rather old bugfix 2.x version from august, see: ender-3-bugfix-2.0.x-skr-e3-dip branch

Directly after boot up:
`
> 11:54:04: N1505 M122 *51
< 11:54:04: X	Y	Z	E
< 11:54:04: Enabled		false	false	false	false
< 11:54:04: Set current	400	400	250	400
< 11:54:04: RMS current	397	397	441	397
< 11:54:04: MAX current	560	560	622	560
< 11:54:04: Run current	12/31	12/31	7/31	12/31
< 11:54:04: Hold current	6/31	6/31	3/31	6/31
< 11:54:04: CS actual	6/31	6/31	31/31	6/31
< 11:54:04: PWM scale	7	7	34	7
< 11:54:04: vsense		1=.18	1=.18	0=.325	1=.18
< 11:54:04: stealthChop	true	true	true	false
< 11:54:04: msteps		256	256	8	256
< 11:54:04: tstep		max	max	max	max
< 11:54:04: pwm
< 11:54:04: threshold				
< 11:54:04: [mm/s]					
< 11:54:04: OT prewarn	false	false	false	false
< 11:54:04: OT prewarn has
< 11:54:04: been triggered	false	false	false	false
< 11:54:04: off time	3	3	3	3
< 11:54:04: blank time	24	24	36	24
< 11:54:04: hysteresis
< 11:54:04: -end		-2	-2	-3	-2
< 11:54:04: -start		6	6	6	6
< 11:54:04: Stallguard thrs				
< 11:54:04: DRVSTATUS	X	Y	Z	E
< 11:54:04: stst		*	*	*	*
< 11:54:04: olb					*
< 11:54:04: ola					*
< 11:54:04: s2gb					
< 11:54:04: s2ga					
< 11:54:04: otpw					
< 11:54:05: ot					
< 11:54:05: 157C					
< 11:54:05: 150C					
< 11:54:05: 143C					
< 11:54:05: 120C					
< 11:54:05: s2vsa					
< 11:54:05: s2vsb					
< 11:54:05: Driver registers:
< 11:54:05: X	0xC0:06:00:00
< 11:54:05: Y	0xC0:06:00:00
< 11:54:05: Z	0xC0:1F:00:00
< 11:54:05: E	0x80:06:00:C0
< 11:54:05: Testing X connection... OK
< 11:54:05: Testing Y connection... OK
< 11:54:05: Testing Z connection... OK
< 11:54:05: Testing E connection... OK
`

After Failsafe setttings:
`
> 11:54:45: N1525 M122 *49
< 11:54:45: X	Y	Z	E
< 11:54:45: Enabled		false	false	false	false
< 11:54:45: Set current	400	400	250	400
< 11:54:45: RMS current	397	397	244	397
< 11:54:45: MAX current	560	560	344	560
< 11:54:45: Run current	12/31	12/31	7/31	12/31
< 11:54:45: Hold current	6/31	6/31	3/31	6/31
< 11:54:45: CS actual	6/31	6/31	3/31	6/31
< 11:54:45: PWM scale	7	7	4	7
< 11:54:45: vsense		1=.18	1=.18	1=.18	1=.18
< 11:54:45: stealthChop	true	true	false	false
< 11:54:45: msteps		256	256	256	256
< 11:54:45: tstep		max	max	max	max
< 11:54:45: pwm
< 11:54:45: threshold				
< 11:54:45: [mm/s]					
< 11:54:45: OT prewarn	false	false	false	false
< 11:54:45: OT prewarn has
< 11:54:45: been triggered	false	false	false	false
< 11:54:45: off time	3	3	3	3
< 11:54:45: blank time	24	24	24	24
< 11:54:45: hysteresis
< 11:54:45: -end		-2	-2	-2	-2
< 11:54:45: -start		6	6	6	6
< 11:54:45: Stallguard thrs				
< 11:54:45: DRVSTATUS	X	Y	Z	E
< 11:54:45: stst		*	*	*	*
< 11:54:45: olb					*
< 11:54:45: ola					*
< 11:54:46: s2gb					
< 11:54:46: s2ga					
< 11:54:46: otpw					
< 11:54:46: ot					
< 11:54:46: 157C					
< 11:54:46: 150C					
< 11:54:46: 143C					
< 11:54:46: 120C					
< 11:54:46: s2vsa					
< 11:54:46: s2vsb					
< 11:54:46: Driver registers:
< 11:54:46: X	0xC0:06:00:00
< 11:54:46: Y	0xC0:06:00:00
< 11:54:46: Z	0x80:03:00:00
< 11:54:46: E	0x80:06:00:C0
< 11:54:46: Testing X connection... OK
< 11:54:46: Testing Y connection... OK
< 11:54:46: Testing Z connection... OK
< 11:54:46: Testing E connection... OK
`

ss89, 2020-01-26:
i don't understand why, but reloading factory settings and saving them does the trick.
Now my only remaining issue is that the steppers are quite noisy

sjasonsmith, 2020-01-26:
The current example has HYBRID_THRESHOLD enabled with pretty low values. You can either disable it completely or increase the threshold where it switches to spread cycle mode.

boelle, 2020-01-28:
will close this as it not a bug as such but rather a config/setup problem
it could be argued that the config has a low value but it will be an individual taste what is low or high

closed 2020-01-28 by boelle