MarlinFirmware/Marlin #16680: [BUG] Cannot enable stealthchop and extruder moves along Y axis.

TheRealVira, 2020-01-26:
### Bug Description

I am trying to get my `bigtreetech skr v1.4` work along with `TMC2209` driver using `UART`. Furthermore, I have followed their guide on their repo: [Manual](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/BTT%20SKR%20V1.4%20Instruction%20Manual.pdf) (cut off the `DIAG` pin since I do not need the `stallguard` function)

However, I cannot seem to get `stealthchop` to work and weirdly enough my extruder moves along the y axis (when it moves up, the extruder starts retracting; moving down, the extruder extrudes). It seems like I am unable to directly interact with my extruder.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4113412/Configuration.zip)

### Steps to Reproduce

1. Clone https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3.git
2. Open `/BIGTREETECH-SKR-V1.3/BTT SKR V1.4/Firmware/Marlin-bugfix-2.0.x-SKR-V1.4` in VS Code e.g.
2. Use dedicated pins accordingly
3. Use given configurations

**Expected behavior:** Have `stealthchop` enabled and an independable moveable Y axis/extruder.

**Actual behavior:** `Stealthchop` is disabled and extruder is depending on the Y axis.

#### Additional Information

M122 output:

```
Send: M122
Recv:           X       Y       Z       E
Recv: Address           0       0       0       0
Recv: Enabled           false   false   false   false
Recv: Set current       725     725     725     725
Recv: RMS current       703     703     703     703
Recv: MAX current       991     991     991     991
Recv: Run current       22/31   22/31   22/31   22/31
Recv: Hold current      11/31   11/31   11/31   11/31
Recv: CS actual 11/31   11/31   11/31   11/31
Recv: PWM scale 13      13      13      13
Recv: vsense            1=.18   1=.18   1=.18   1=.18
Recv: stealthChop       false   false   false   false
Recv: msteps            16      16      16      16
Recv: tstep             max     max     max     max
Recv: pwm
Recv: threshold
Recv: [mm/s]
Recv: OT prewarn        false   false   false   false
Recv: off time  3       3       3       3
Recv: blank time        24      24      24      24
Recv: hysteresis
Recv: -end              -1      -1      -1      -1
Recv: -start            1       1       1       1
Recv: Stallguard thrs   0       0       0       0
Recv: DRVSTATUS X       Y       Z       E
Recv: stst              *       *       *       *
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv:           X       0x80:0B:00:00
Recv:           Y       0x80:0B:00:00
Recv:           Z       0x80:0B:00:00
Recv:           E       0x80:0B:00:00
Recv:
Recv:
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: ok
```

ellensp, 2020-01-26:
Unfortunately  https://github.com/bigtreetech is not a The official Marlin firmware repository. This repository is controlled and maintained by bigtreetech and you have to direct support request to them about it.

Alternatively download the official Marlin from https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x and try that. If you still have issue then we can help.

TheRealVira, 2020-01-26:
> Unfortunately https://github.com/bigtreetech is not a The official Marlin firmware repository. This repository is controlled and maintained by bigtreetech and you have to direct support request to [...]

So, apparently this at least fixed the stealthchop mode! However, the extruder still seems to be dependent on the Y axis.. :pensive:

boelle, 2020-01-26:
closing

as @ellensp says that github repro is not ours and we dont support the others

closed 2020-01-26 by boelle

TheRealVira, 2020-01-26:
It has the same behavior though (see my most recent reply in this ticket).

boelle, 2020-01-26:
i have stealthchop on all axis and extruder, my Y axis does not move when extruder moves

and extruder does not move when y axis move

TheRealVira, 2020-01-26:
boelle, thank you for your time, but please read my ticket more carefully..

Part of my issue is exactly that my extruder only moves when my Y stepper motor is moving. (Printhead moves up and Extruder retracts based on its y position; printhead moves down and extruder extrudes based on its y position.) I cannot move the extruder independently!

TheRealVira, 2020-01-26:
I want extruder not to move because of Y axis movement.

boelle, 2020-01-26:
i cant make the same error, i knew what you mean

my extruder moves fine independently
and my y axis moves independently too

TheRealVira, 2020-01-26:
Where is this coming from then?

boelle, 2020-01-26:
some sort of fault on the board

its not UART since that only tell what amp's etc to use, not when to move etc

TheRealVira, 2020-01-26:
Weird.. I guess I should contact BigTreeTech then? I mean, since you have tested it on your board and cannot reproduce said error.

boelle, 2020-01-26:
yes, you paid them and they should give you help and support