MarlinFirmware/Marlin #16681: [FR] LCD menu option for auto extruder fan speed adjust

m00anj00, 2020-01-26:
Feature request:
Hi
would it be possible to have either an M/G command or an LCD menu option to adjust and save auto extruder fan speed ?
The dilemma I have is that using a diamond hotend with auto fan causes me to have a lot of recompile work to find the perfect value.  
For anyone not in the know about diamond hotends:

If the cooler fan is too slow you get blockages in the upper end of the throat tubes.  
Inversely, if the fan is too fast you get blockages in the lower end of the throat tubes and heating issues at worst case due to cooling the top edge of the heater itself as they are longer than the diamond hot end.

I would assume that there is benefit also for people using fans on direct drive set ups as the noise is annoying, and throttling the fan is preferred, but if too slow you will end up with a silly string mess in the extruder drive due to heat travelling from the throat and heat generated by the stepper.

Any opinions on this welcome

Thankd

ellensp, 2020-01-26:
In the mean time, E0_AUTO_FAN_PIN is listed as a protected pin. If you remove this from sensitive_pins.h  you can use the gcode M42 P{your auto fan pin} S{PWM speed} to adjust the speed, but only after marlin has already turned on the fan.

m00anj00, 2020-01-26:
Thanks.  That will work.  Would be nice to be able to 'on the fly' set it
but keep the automated fan (which is great btw).  :)