MarlinFirmware/Marlin #16683: [BUG] Sudden temprature drops while printing

opiswahn, 2020-01-26:
### Bug Description

Since today, i got some wired temp drops while printing, not time related
[image]
That is a scale from about 30min. Sometimes the drops happen after 5min, sometimes after 20min or later, also regardless of model that was printed.
I am using the current bugfix-2.0.x from today, my older version that worked was from around 15 days ago.
It has nothing to do with part cooling fan turning on, that one is running since 2nd layer.

The temp is pretty stable until it drops, max 1-2 degrees delta in a very short timespan(1-5 sec). Those sudden spikes down last for some 30sec-1min, until my runaway-protection kicks in.

I never had something like this before, some little dropy yes, but mot on that scale...

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4114310/Marlin.zip)

### Steps to Reproduce

1. Print
2. Wait for temp drops.

GMagician, 2020-01-26:
Check the cable. I had some issue like that and was bed cable broken, you can check it with a voltmeter.

opiswahn, 2020-01-26:
Cable is fine, a already checked that.

boelle, 2020-01-26:
bad thermistor maybe, or close to going bad

yet-another-average-joe, 2020-01-26:
I had the exact same issue with a genuine brand new E3D V6 hotend, because of the thermistor connector ; the connector itself was fine, but they do crimp the terminal pins to the thermistor solid wires. This is nonsense, crimping is for stranded wires, never for solid wires. As a result, I had intermittent contact, and from time to time thermal runaway errors. The thermistor being a safety device, I replaced it with another one, and connected its 1 meter wires directly to the motherboard.

So, if you have such a connector near to the thermistor, have a look to it...

Where I live, crimping ferrules on solid wires is prohibited.

thinkyhead, 2020-01-26:
Marlin is trusting what the sensors tell it and applying PID in classic manner. Marlin is not trying to be too clever, and certainly it is not trying to be clever by turning off the heaters. I've been doing multi-hour prints on several machines with recent and current Marlin 2.0.x and we have not seen any issues with the firmware deciding to stop heating.

yet-another-average-joe, 2020-01-27:
> 
> 
> ... Marlin is not trying to be too clever, and certainly it is not trying to be clever by turning off the heaters...
>

@thinkyhead , I don't understand...
Attempting to heat the hotend without the thermistor, gives "Exx heating failed" error (heater does not turn on), and Marlin stops, turning the heaters off ; if the temp reported by the thermistor drops too much, too long, or is too unstable (or if the sensor is disconnected too long while heating), it does the same (thermal runaway error + turns off the heaters). Marlin is clever and safe !

OscarGciam, 2020-01-27:
Same problem with the same board (SKR 1.3) over here with Marlin 2.0.1

boelle, 2020-01-28:
> Cable is fine, a already checked that.

also check the strands are not broken inside the cable

italocjs, 2020-01-30:
I'll leave one printer with a new e3d (which has never seen plastic) hot tomorrow for a few hours to help testing this, i have printer a few hours today (with today's bugfix) and didn't notice any issue on the printed part.

italocjs, 2020-01-31:
i had no issues so far, as said in the previous comment i burned yesterday bugfix, did the pid autotune and turned the hotend on (without any plastic inside) for around 4 hours, there were no spikes or drops in the temperature (using 260C - cloned e3dv6, 100k ntc termistor)

opiswahn, 2020-01-31:
I'll try to replace the temp probe and heating cartridge this weekend(together with cables) and see if that has any effekt.

opiswahn, 2020-01-31:
I don't think it is the thermistor, if that would malefunction, the temp would go to 0 and back and not gradually drop.

italocjs, 2020-01-31:
About the termistor, it doesn't have to fully short or break the connection, just enough to mess with the resistance. I've switched a few thermistor so far, all reporting weird temperatures

Obter o Outlook para Android<https://aka.ms/ghei36>

houseofbugs, 2020-02-01:
Cannot replicate on any versions of Marlin. This is a hardware issue, not a Marlin bug. Could be your board or the thermistor (wiring included). Replace the thermistor and associated wiring. If that doesnt show any change then swap the control board.

boelle, 2020-02-01:
will close this one for the reasons @houseofbugs just mentioned

we can always reopen if needed

closed 2020-02-01 by boelle